OnionIoT/OpenWRT-Packages #4: nano not working

edent, 2016-12-24:
Not sure if this is a problem here or upstream.

```
root@Omega-1234:~# opkg update
root@Omega-1234:~# opkg install nano
root@Omega-1234:~# nano
Error loading shared library libncursesw.so.6: No such file or directory (needed by /usr/bin/nano)
Error relocating /usr/bin/nano: raw: symbol not found
Error relocating /usr/bin/nano: napms: symbol not found
Error relocating /usr/bin/nano: key_defined: symbol not found
Error relocating /usr/bin/nano: doupdate: symbol not found
Error relocating /usr/bin/nano: nodelay: symbol not found
Error relocating /usr/bin/nano: waddch: symbol not found
Error relocating /usr/bin/nano: wscrl: symbol not found
Error relocating /usr/bin/nano: curs_set: symbol not found
Error relocating /usr/bin/nano: waddnstr: symbol not found
Error relocating /usr/bin/nano: wattr_off: symbol not found
Error relocating /usr/bin/nano: wmove: symbol not found
Error relocating /usr/bin/nano: wrefresh: symbol not found
Error relocating /usr/bin/nano: scrollok: symbol not found
Error relocating /usr/bin/nano: beep: symbol not found
Error relocating /usr/bin/nano: initscr: symbol not found
Error relocating /usr/bin/nano: nonl: symbol not found
Error relocating /usr/bin/nano: tigetstr: symbol not found
Error relocating /usr/bin/nano: wnoutrefresh: symbol not found
Error relocating /usr/bin/nano: isendwin: symbol not found
Error relocating /usr/bin/nano: noecho: symbol not found
Error relocating /usr/bin/nano: wgetch: symbol not found
Error relocating /usr/bin/nano: delwin: symbol not found
Error relocating /usr/bin/nano: wattr_on: symbol not found
Error relocating /usr/bin/nano: newwin: symbol not found
Error relocating /usr/bin/nano: set_escdelay: symbol not found
Error relocating /usr/bin/nano: keypad: symbol not found
Error relocating /usr/bin/nano: endwin: symbol not found
Error relocating /usr/bin/nano: LINES: symbol not found
Error relocating /usr/bin/nano: curscr: symbol not found
Error relocating /usr/bin/nano: stdscr: symbol not found
Error relocating /usr/bin/nano: COLS: symbol not found
```

If I try to manually install `libncursesw` I get

```
root@Omega-1234:~# opkg install libncursesw
Package libncursesw (5.9-4) installed in root is up to date.
```

WereCatf, 2016-12-24:
Upstream changed the library-name to libncurses for compatibility-reasons. The Onion-devs should update their images and build against libncurses, not libncursesw.

Uncomment the reboot_luci-package-source in /etc/opkg/ and install libncurses with `opkg --force-overwrite install libncurses` for now, if you want nano working.

edent, 2016-12-25:
Perfect! 

For those following along at home, here's what I did,

```
echo 'src/gz reboot_luci http://downloads.lede-project.org/snapshots/packages/mipsel_24kc/luci' >> /etc/opkg/distfeeds.conf
opkg -update
opkg --force-overwrite install libncurses
```

Nano now works. Christmas is saved!

greenbreakfast, 2017-01-06:
This should be fixed on the Omega as of firmware version 0.1.7 (with no manual changes to the `/etc/opkg/distfeeds.conf` file)

Let us know if it doesn't work!

closed 2017-01-06 by greenbreakfast