OnionIoT/OpenWRT-Packages #13: git@github.com:OnionIoT/onion-st7735-driver.git cannot be accessed

HariHend1973, 2018-02-27:
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

kabongsteve, 2018-08-26:
I'd like the same package, but same issue
Seems no-one is interested in responding either

greenbreakfast, 2018-08-31:
This was a package we created to test ST7735 displays with the original Omega. The code was never properly finished and is absolutely full of bugs.
Are you guys trying to use the Omega with ST7735 displays? If so, I would recommend using our [python-spidev](https://github.com/OnionIoT/python-spidev) package and seeing if you can create your own Python program.

greenbreakfast, 2018-09-12:
As mentioned in #15, maybe a framebuffer device would be a good solution? Take a look at our community threads: 
http://community.onion.io/topic/3056/omega2-omega2plus-image-build-process-for-fbtft-frame-buffer-tftlcd-devices
http://community.onion.io/topic/3050/fb_ili9341-80fps-up-to-100fps-memcpy-387-fps-minimal

Let's move the deprecated package conversation to issue #5 since it involves the same thing.

closed 2022-11-30 by greenbreakfast