OnionIoT/OpenWRT-Packages #14: Wifimanager does not understand emoji SSID's

ruudk, 2018-06-05:
I try to connect to my WiFi network with SSID "💩"

But for some reason, it doesn't work.

When I run wifisetup it shows the UTF-8 char as HEX, see:

```
$ wifisetup
Onion Omega Wifi Setup

Select from the following:
1) Scan for Wifi networks
2) Type network info
q) Exit

Selection: 1
Scanning for wifi networks...

Select Wifi network:
1) 0xF09F92A9
2) 0xF09F92A9
```

When I just leave it in hex and connect, it doesn't work. When I manually enter the poop emoji it doesn't work either.

Does anybody have an idea where to start to fix this issue? Does the wifimanager not support UTF-8 or is this something deeper?

greenbreakfast, 2018-06-06:
It's something deeper, the wireless driver does not support unicode characters at this time.

ruudk, 2018-06-07:
Do you know where I can report the issue to get this resolved?

closed 2019-11-02 by ruudk