OnionIoT/OpenWRT-Packages #15: Git repo for the ili9341 does not exist (anymore)

eflukx, 2018-06-20:
https://github.com/OnionIoT/OpenWRT-Packages/blob/28127d8457cbf76cfd4e924c6feb03ec7f1f48c9/onion-ili9341-driver/Makefile#L18

greenbreakfast, 2018-08-31:
This was a repo for testing ILI9341 displays with a C-based driver. It was never completed since we discovered the C library we were creating did not suit general purpose use of the screen.

If you're interested in using an ILI9341 display with the Omega, I would recommend using our [python-spidev](https://github.com/OnionIoT/python-spidev) package to create a Python program to control your display.

eflukx, 2018-08-31:
Oj.. thats a pity. Using the ili9341 using python wont cut it, both in terms of performance and usability (i.e. fbdev device). 

I any way, the stale reference to a non-existing repo should be removed imo (breaks the build).

Cheers!

eflukx, 2018-09-10:
If this is deprecated (which makes sense when code is unfinished/unreliable!), please remove the package definition, as the only thing it currently does is braking the build.

See also issue #13

greenbreakfast, 2018-09-12:
You'll be happy to see that our community has been experimenting with fbdev extensively:
http://community.onion.io/topic/3056/omega2-omega2plus-image-build-process-for-fbtft-frame-buffer-tftlcd-devices
http://community.onion.io/topic/3050/fb_ili9341-80fps-up-to-100fps-memcpy-387-fps-minimal

Let's move the deprecated package conversation to issue #5 since it involves the same thing.

closed 2018-09-12 by eflukx