OnionIoT/OpenWRT-Packages #17: onion os does not work under https

dlparker, 2018-11-07:
Not sure if this is the right github project, but it is the only one I can find that has onion os code.

This page https://onion.io/2bt-https-web-server/ shows how to setup https for the onion os web server, but the result does not work under chrome. I can get the login page to show up by tweaking settings, but the login doesn't actually work. 

There are at least two problems blocking this. First, the page tries to open a websocket back to the server, but uses ws://192.168.3.1 when it should use wss://192.168.3.1, so chrome blocks it. 

Second, it tries to get resources from a CDN at http://s3-us-west-2.amazonaws.com/onion-cdn/os/GA/index.html which is http instead of the requires https, so chrome blocks it. 

Seems like these should be addressed, or the article should be removed or edited.

greenbreakfast, 2018-11-07:
Yeah, you're right, OnionOS isn't 100% ready for use with HTTPS. We'll add that to the backlog.

However, this article is still useful if you're running other things on the Omega's web-server and would like to secure it with HTTPS.

dlparker, 2018-11-07:
Ok, that makes sense, but I still think the article will mislead people. I figured it out pretty fast, but I am used to debugging web apps. If you weren't then you might give up in frustration thinking it doesn't work. 

A update at the end of the article to explain that you can't actually log in but '... what you said ...' would be good.

closed 2022-11-30 by greenbreakfast