OnionIoT/OpenWRT-Packages #19: Disappearing of default gateway setting

mverlaan, 2018-11-21:
Hi,

I’m experiencing an issue with the Omega 2+ board.
When I set the default gateway on apcli0, it automatically gets removed from the interface.

I did some debugging myself, to my knowledge this is caused by the ap_client tool. The ap_client tool uses the calls: ifconfig apcli0 down; ifconfig apcli0 up. The call ifconfig down will make the kernel forget the gateway entry and ifconfig up does not restore this.

To my understanding this can be fixed by using the calls: ifdown and ifup instead of ifconfig(or using an ubus call to refresh the network configuration).

Because the ap_client tool is only a binary in the repo, I can’t fix this myself.

Any help is much appreciated.

mverlaan, 2018-11-22:
Hi, thanks for the quick response.

I'm trying to connect the client on the Onion (wwan) to a wireless router which is not running a DHCP service. The wireless router will be set as the gateway setting for the Onion. My experience is that when I configure the network settings of the Onion using DHCP, the gateway setting sticks, but when I configure it statically, it disappears. I wrote a script that will poke the static network settings into openWRT using UCI. Currently I'm using the 0.2.0-b188 software version.

I'm using the following script:
`if [ ${valid} -ne 0 ] && AskAcceptance
	then
		echo "Applying settings..."
 
		if [ ${enabled} -ne 0 ]
		then
			uci set wireless.sta.disabled=0
			uci set wireless.device='wifi-config'
			uci set wireless.device.key="${key}"
			uci set wireless.device.ssid="${ssid}"
			uci set wireless.device.encryption='psk2'
 
		if [ ${dhcp} -ne 0 ]
		then
			uci set network.wwan.proto='dhcp'
			uci -q delete network.wwan.ipaddr
			uci -q delete network.wwan.netmask
			uci -q delete network.wwan.gateway
			uci -q delete network.wwan.dns
		else
			uci set network.wwan.proto='static'
			uci set network.wwan.ipaddr="${ipaddr}"
			uci set network.wwan.netmask="${netmask}"
			if [ -n "${gateway}" ]
			then
				uci set network.wwan.gateway="${gateway}"
			else
				uci -q delete network.wwan.gateway
			fi
			if [ -n "${dns}" ]
			then
				uci set network.wwan.dns="${dns}"
			else
				uci -q delete network.wwan.dns
			fi
		fi
	else
		uci set wireless.sta.disabled=1
	fi

	uci commit
	UbusInvokeEvent 'wireless'
	UbusInvokeEvent 'network'
else
	echo "Discarding settings..."
fi

#
# Function to invoke a config.change ubus call
#
UbusInvokeEvent ()
{
	local package="$1"
 
	ubus call service event "{ \"type\": \"config.change\", \"data\": { \"package\": \"${package}\" }}"
}`

Thanks.

greenbreakfast, 2018-11-22:
Can you provide examples of what your `network` and `network` files from `/etc/config` look like for each of your two cases (with dhcp and without dhcp)
Will make it easier to help

mverlaan, 2018-11-23:
Hi,

Thanks for the response, I put the config files at the end if this comment.

The steps you can follow to reproduce the problem:
1. Setup a wireless access point
2. Power down the wireless access point(it should be unavailable)
3. Configure the Onion wireless network using the following commands:
```
uci set network.wwan=interface
uci set network.wwan.ifname='apcli0'
uci set network.wwan.proto='static'
uci set network.wwan.ipaddr='192.168.1.130'
uci set network.wwan.netmask='255.255.255.0'
uci set network.wwan.gateway='192.168.1.1'
uci commit
reload_config
```
	
4. Check the settings by using: uci show network
5. Check with "route" if there is a gateway set, it should;
```
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.1.1     0.0.0.0         UG    0      0      0   apcli0
```
6. Power up the wireless access point
7. Check the gateway setting with the "route" command again, it should have disappeared.
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref     Use Iface
192.168.1.0     *               255.255.255.0   U     0      0       0   apcli0
```

8. Power down the wireless access point, the gateway setting should appear again:
```
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.1.1     0.0.0.0         UG    0      0      0   apcli0
```

9. You can do the same steps with a dhcp configuration:
```
uci set network.wwan=interface
uci set network.wwan.ifname='apcli0'
uci set network.wwan.proto='dhcp'
uci commit
reload_config
```
10. If you power up the wireless access point, the Onion will keep its default gateway setting

I hope you have the opportunity to reproduce this yourself.

```
======= network config with dhcp ===========
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.wlan=interface
network.wlan.type='bridge'
network.wlan.proto='static'
network.wlan.ipaddr='169.254.0.1'
network.wlan.netmask='255.255.0.0'
network.wan=interface
network.wan.ifname='eth0'
network.wan.proto='dhcp'
network.wwan=interface
network.wwan.ifname='apcli0'
network.wwan.proto='dhcp'

Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric    Ref      Use   Iface
default         192.168.1.1     0.0.0.0         UG    0         0        0     apcli0
169.254.0.0     *               255.255.0.0     U     0         0        0     br-wlan
192.168.1.0     *               255.255.255.0   U     0         0        0     apcli0
192.168.1.1     *               255.255.255.255 UH    0         0        0     apcli0

======= network config with static ===========
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.wlan=interface
network.wlan.type='bridge'
network.wlan.proto='static'
network.wlan.ipaddr='169.254.0.1'
network.wlan.netmask='255.255.0.0'
network.wan=interface
network.wan.ifname='eth0'
network.wan.proto='dhcp'
network.wwan=interface
network.wwan.ifname='apcli0'
network.wwan.proto='static'
network.wwan.ipaddr='192.168.1.130'
network.wwan.netmask='255.255.255.0'
network.wwan.gateway='192.168.1.1

Kernel IP routing table
Destination     Gateway         Genmask         Flags   Metric    Ref      Use    Iface
169.254.0.0     *               255.255.0.0     U       0         0        0      br-wlan
192.168.1.0     *               255.255.255.0   U       0         0        0      apcli0

======== /etc/config/wireless ============
wireless.radio0=wifi-device
wireless.radio0.type='ralink'
wireless.radio0.variant='mt7628'
wireless.radio0.country='US'
wireless.radio0.hwmode='11g'
wireless.radio0.htmode='HT40'
wireless.radio0.channel='auto'
wireless.radio0.disabled='0'
wireless.radio0.device_mode='apsta'
wireless.radio0.op_mode='preference'
wireless.ap=wifi-iface
wireless.ap.device='radio0'
wireless.ap.mode='ap'
wireless.ap.network='wlan'
wireless.ap.ifname='ra0'
wireless.ap.encryption='psk2'
wireless.ap.disabled='0'
wireless.ap.key='wifi_device'
wireless.ap.ssid='device-00002'
wireless.sta=wifi-iface
wireless.sta.device='radio0'
wireless.sta.mode='sta'
wireless.sta.ifname='apcli0'
wireless.sta.encryption='psk2'
wireless.sta.network='wwan'
wireless.sta.disabled='0'
wireless.sta.key='device_pass'
wireless.sta.ssid='wirelessAP'
wireless.device=wifi-config
wireless.device.encryption='psk2'
wireless.device.key='device_pass'
wireless.device.ssid='wirelessAP'
```

mverlaan, 2018-12-05:
Have you been able to reproduce the problem with the information I provided?