OnionIoT/OpenWRT-Packages #33: php7-cli not longer available

jjok, 2020-03-06:
Hi. I used to have `php-cli` installed, when it was PHP 7.0 (I think. Or was it 7.1?). Since PHP 7.2 has been added, `cli` is not longer available. 

> You must actually choose your PHP flavour (cli, cgi or fastcgi)

```
opkg update
opkg list | grep php
```
```
php7 - 7.2.26-1 - PHP is a widely-used general-purpose scripting language that is especially suited for Web development and can be embedded into HTML. This package contains only the PHP config file. You must actually choose your PHP flavour (cli, cgi or fastcgi).  Please note, that installing php5 and php7 in parallel on the same target is not supported in OpenWrt/LEDE.
php7 - 7.2.25-1
php7-cgi - 7.2.25-1
php7-cgi - 7.2.26-1 - PHP is a widely-used general-purpose scripting language that is especially suited for Web development and can be embedded into HTML. This package contains the CGI version of the PHP7 interpreter.
php7-fastcgi - 7.2.26-1 - As FastCGI support is now a core feature the php7-fastcgi package now depends on the php7-cgi package, containing just the startup script.
...
```

greenbreakfast, 2020-03-06:
Try enabling the openwrt package repos to install php-cli. 

By default, the opkg package manager on the Omega2 platform is configured use Onion’s online package repositories. We host a selection of packages that are frequently used by the Omega2 userbase. However, the official OpenWRT repositories host a larger selection of packages.

If you’re looking for a specific package and it isn’t available by default, you can try enabling the official OpenWRT package repositories on your Omega.

Learn more here: http://docs.onion.io/omega2-docs/using-opkg.html#you-dont-have-a-package-i-wantneed

(Responded to the [thread on Onion community](https://community.onion.io/topic/4065/resolved-php-cli-can-no-longer-be-installed/3) as well)

closed 2020-03-06 by greenbreakfast

jjok, 2020-03-07:
Thanks. I don't think I've seen that section of the docs before.

I thought it was strange that `php7-cgi` and `php7-fastcgi` were included, but not `php7-cli`. I don't remember having to enable additional repos in the old version.