OnionIoT/OpenWRT-Packages #44: modules like kmod-wireguard don't install - kernel version mismatch

CRImier, 2022-06-16:
``` bash
# opkg install kmod-wireguard
Installing kmod-wireguard (4.14.81+0.0.20180718-2) to root...
Downloading http://repo.onioniot.com/omega2/packages/core/kmod-wireguard_4.14.81%2b0.0.20180718-2_mipsel_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-wireguard:
 *      kernel (= 4.14.81-1-d38bb0cccb6c2f5c91562ebf1fb26fa2) *         kernel (= 4.14.81-1-d38bb0cccb6c2f5c91562ebf1fb26fa2) *         kernel (= 4.14.81-1-d38bb0cccb6c2f5c91562ebf1fb26fa2) * 
 * opkg_install_cmd: Cannot install package kmod-wireguard.
```

opkg list entries:
```
kernel - 4.14.81-1-10b6683764d58480e54ef73aa783b9bb
[...]
kmod-wireguard - 4.14.81+0.0.20180718-2 - WireGuard is a novel VPN that runs inside the Linux Kernel and utilizes state-of-the-art cryptography. It aims to be faster, simpler, leaner, and more useful than IPSec, while avoiding the massive headache. It intends to be considerably more performant than OpenVPN.  WireGuard is designed as a general purpose VPN for running on embedded interfaces and super computers alike, fit for many different circumstances. It uses UDP.  This package provides the kernel module for WireGuard.
```
quite weird yeah
I have all upstream openwrt distfeed enabled except core

greenbreakfast, 2022-06-16:
Two suggestions:

1. You'll want to disable the upstream openwrt dist feeds and rely on the onion feeds instead - the kernel checksum of the kernel modules in the Onion feeds will match that of the Onion firmware
2. Make sure you're running the very latest firmware build for your Omega2/Omega2+, again to match the kernel checksum of the kernel packages from the Onion feed to that of the firmware running on your device

More info [here](http://docs.onion.io/omega2-docs/using-opkg.html#onion-package-repo)

CRImier, 2022-06-16:
`oupgrade` was done yesterday or so
downloading a `kernel` ipk from onion repo and installing it seems to have solved this?
at that, it seems I might've downloaded the wrong one on accident, with a different hash, and it *still* helped? can no longer verify that part, sadly
which is weird, because I did try disabling the upstream repos before and it didn't work, and I'm pretty sure I didn't install an upstream kernel
ohwell =D

closed 2022-06-16 by CRImier