OnionIoT/OpenWRT-Packages #46: Ability to control GPIO mux

greenbreakfast, 2023-02-17:
# Motivation

Need a method to control the GPIO muxing of Omega2 in OpenWRT 22.03-based firmware.
GPIO muxing allows users to change the functionality of the pins from the system defaults, enabling more use cases.
[image]

# Description

A new package is available in the Onion package repo and bundled in the firmware. This package provides a command line utility which users can run to change the GPIO muxing.

Example commands:

To get the current mode of the Omega’s multiplexed pins, use the command:
```
omega2-ctrl gpiomux get
```

To set a particular group of hardware pins to a specified mode, use the following command:

```
omega2-ctrl gpiomux set <HARDWARE PIN GROUP> <MODE>
```

For example, the following command will set UART1 pins to operate in GPIO mode:
```
omega2-ctrl gpiomux set uart1 gpio
```

# Alternatives Considered

Using `/dev/mem` and writing GPIO mux to register settings directly is a potential alternative. However, this is prone to errors and a difficult user experience.

# Additional Resources

`omega2-ctrl` is a C program that can perform the required GPIO muxing actions, and works on OpenWRT 18.06. The source code can be found here: https://github.com/OnionIoT/omega2-ctrl

And the package makefile can be found here: https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-18.06/omega2-ctrl/Makefile

**This enhancement focuses on porting the `omega2-ctrl` package to OpenWRT 22.03**

# Reference to existing IPK package file?

See above

jempatel, 2023-03-17:
@greenbreakfast Do we have any reference documentation that I can test with hardware reference and its outcome?

jempatel, 2023-03-17:
I was able to compile the package successfully, but to make it work we need `/dev/mem`, which can be enabled by adding `CONFIG_KERNEL_DEVMEM' into `.config`.

On the console, it looks like this

[image]

greenbreakfast, 2023-03-17:
Accidentally closed the issue - reopening it

We've shown this is doable, let's integrate it into the firmware. 
Depends on https://github.com/OnionIoT/openwrt-imagebuilder-wrapper/issues/4 being completed before firmware integration is possible

DocHardinger, 2023-03-25:
Hi,

first great that youre working on OpenWrt 22.03. Thats really satisfying also the new Buildsystem wrapper.
I want to test the omega2-ctrl on OpenWrt 22.03 with your custom 22.03. image. 
I tried to use your openwrt-sdk-wrapper, but seems some files for omega2-ctrl are missing in the 22.03 tag package repo (https://github.com/OnionIoT/OpenWRT-Packages/tree/openwrt-22.03/omega2-ctrl) .
So how can i build the omega2-ctrl package? thx

greenbreakfast, 2023-03-27:
Hi @DocHardinger, happy to hear you're just as excited about OpenWrt 22.03 as we are :)

**Short answer** to your omega2-ctrl package question: you can't build it quite yet, but we're actively working on it

**The long answer:**
Currently the [beta firmware we mention on the Onion Community](https://community.onion.io/topic/4963/beta-openwrt-22-03-firmware-available-your-feedback-is-needed) is being built using the OpenWRT [SDK + openwrt-sdk-wrapper](https://github.com/OnionIoT/openwrt-sdk-wrapper) and the [Image builder +  openwrt-imagebuilder-wrapper](https://github.com/OnionIoT/openwrt-imagebuilder-wrapper). This is a much, much quicker build process than compiling everything from scratch using the openwrt build system. 

However, this limits us to the kernel configuration that's set in the OpenWRT SDK published by the OpenWRT community.

The rub is the omega2-ctrl package depends on the kernel being configured to expose the devmem interface to userspace. 

So, we are in the process of moving the openwrt-sdk-wrapper and openwrt-imagebuilder-wrapper to use Onion-customized versions of the SDK and Image Builder (which expose the devmem interface). 

You can track our progress here: https://github.com/OnionIoT/openwrt-imagebuilder-wrapper/issues/4

And the dependencies we have to complete first:
* https://github.com/OnionIoT/openwrt-imagebuilder-wrapper/issues/3
* https://github.com/OnionIoT/openwrt-buildsystem-wrapper/issues/7

DocHardinger, 2023-03-27:
Ah Ok. Thx for the reply @greenbreakfast 
I already track your progress. I check it every day ;-).
We´re inventing a smart van system (see http://e1nfach.de , for now only in germany) and decided to use the omega2 as the master brain. 
Before, i tested the Omega with 22.03. but only for all software stuff like Modem, Routing, python3 and so on. Works all well.
But now we discovered that Hardware IO, PWM and other stuff isnt working quite well. Then i tried the old 18.06. firmware but here usb qmi modems are not working because of some pakages (usbnet, etc), and also other stuff is not as nice as in 22.03. .

Any ideas which firmware to use for working software stuff (escpecially quectel qmi modem, python3, routing stuff, mqtt) AND Hardware GPIO,PWM,I2C,etc? We can also use the Firmware/package Builder if neccessary. I tried a lot different options on building but no success.

Would be great (as long as youre working on the 22.03. firmware) to find a solution here so we can continue our project.
Were not really into C or C++ otherwise we would help you :-)

By the way, do you know if the omega2 lte (global version OM-O2LTE-G) will have a come back to the european market? 

Thx for your time

greenbreakfast, 2023-03-29:
@DocHardinger very cool product, I love the niche! I'm stoked you've chosen the Omega2 to build around. 

> Before, i tested the Omega with 22.03. but only for all software stuff like Modem, Routing, python3 and so on. Works all well.
> But now we discovered that Hardware IO, PWM and other stuff isnt [...]

This is exactly what we're aiming to solve with the [new firmware effort I posted about on the Onion community](https://community.onion.io/topic/4963/beta-openwrt-22-03-firmware-available-your-feedback-is-needed/2) - bringing utilities + features that add value to Omega2 users to the latest OpenWRT releases.

> Any ideas which firmware to use for working software stuff (escpecially quectel qmi modem, python3, routing stuff, mqtt) AND Hardware GPIO,PWM,I2C,etc?

I'm biased, but I think Onion's version of OpenWRT 22.03 firmware will be your best bet. You get the qmi modem, python3, etc out of the box, and we're actively working on improving access to the hardware interfaces. 

Exact timelines are fuzzy but we're working on it continuously - and as you know, you can track our progress [right here on github](https://github.com/orgs/OnionIoT/projects/1).

You can build your own versions of the firmware we release using our [Image Builder wrapper](https://github.com/OnionIoT/openwrt-imagebuilder-wrapper). That will probably be the easiest way to keep up with our development.

> Would be great (as long as youre working on the 22.03. firmware) to find a solution here so we can continue our project.
> Were not really into C or C++ otherwise we would help you :-)

You can help by letting us know what features/functionality matter most to you! 
The best way to do that is to submit a [new issue in this repo using the "Improvement Suggestion/Feature Request" template](https://github.com/OnionIoT/OpenWRT-Packages/issues/new/choose).

greenbreakfast, 2023-05-17:
omega2-ctrl can now be used with the beta Omega2 firmware!

Download the latest available beta firmware: http://repo.onioniot.com/omega2/images/openwrt-22.03/onion_omega2p-22.03.3-20230517.bin
Install it in your device and the `omega2-ctrl` utility is built in. 

Note: this package relies on changes to the kernel config, so this will only work on the latest (and all future versions) of the beta firmware.

It's also now possible to build your own firmware that is compatible with the `omega2-ctrl` utility

---

Ping @DocHardinger - try it out and reopen this issue if you hit any roadblocks

closed 2023-05-17 by greenbreakfast