OnionIoT/OpenWRT-Packages #47: GPIO Number mapping utility

greenbreakfast, 2023-02-17:
# Motivation

As of Linux kernel 5.10, the GPIO pins are automatically assigned numbers that do not match the Omega2's GPIO numbering. This makes it difficult to use GPIOs

# Description

## Current State

GPIOs 0 - 31 are numbered GPIOs 480 - 511
GPIOs 32 - 63 are numbered GPIOs 448 - 479

## Desired State

Firmware contains a command-line utility that, given an "actual" GPIO number, outputs the equivalent kernel GPIO number.
Utility should be a simple shell script.

For example:

```
# gpio-lookup 15
495

# gpio-lookup 62
478

# gpio-lookup 99
-1 
<Program returns error>
```

# Alternatives Considered

A table published online that maps "actual" GPIO numbers to the kernel GPIO number.
This should be included in the documentation for this package

# Additional Resources

Post on Onion Community by luz identifying the issue: https://community.onion.io/topic/4531/upgrade-to-openwrt-21-02-coming-soon/16

# Reference to existing IPK package file?

None

jempatel, 2023-03-17:
@greenbreakfast  Where can I find the source code of `gpio-lookup` utility?

jempatel, 2023-03-18:
@greenbreakfast got it. I will add a utility for the same.

jempatel, 2023-03-18:
@greenbreakfast  This is a very tiny script. Would you want to have it in a separate package or we can make it part of some existing packages?

Here is a Screenshot.
[image]

Here is a script

```sh
#!/bin/sh

usage() {
	echo "$0 <onion gpio>"
}

if [ $# -lt 1 ]; then
	usage
	exit 1
fi

GPIO_INPUT=$1
GPIO_MIN=0
GPIO_MAX=63
ARCH_NR_GPIOS=512

if ! validate_data "range($GPIO_MIN, $GPIO_MAX)" "$GPIO_INPUT" 2>/dev/null; then
	echo "-1"
	exit 1
fi

GPIO_CHIP=$(( $(( GPIO_INPUT / 32 )) + 1 ))
GPIO_CHIP_BASE=$(( ARCH_NR_GPIOS - $(( 32 * GPIO_CHIP )) ))
GPIO_OFFSET=$(( GPIO_INPUT % 32 ))

GPIO=$(( GPIO_CHIP_BASE + GPIO_OFFSET ))

gpio_sysfs="/sys/class/gpio/gpiochip${GPIO_CHIP_BASE}"
if [ ! -d "$gpio_sysfs" ]; then
	echo "gpio not found"
	exit 1
fi

echo "$GPIO"

```

greenbreakfast, 2023-03-30:
Tested on device, works great, thanks!
Closing this issue

closed 2023-03-30 by greenbreakfast