OnionIoT/OpenWRT-Packages #48: Support NodeJS v16.19.0

greenbreakfast, 2023-02-17:
# Goal

Get latest NodeJS (v16.19.0) package compiled and running for the Omega2 on OpenWRT 22.03

Or: understand exactly why we can’t build

# Motivation

Supporting modern NodeJS opens the Omega2 platform to more developers and more use cases.
It also enables running the latest versions of interesting packages like Node-RED.

The OpenWRT 18.06 Omega2 firmware supports Node v8.10 - which was released back in 2018. Node has evolved a lot since then and these improvements should be brought to the Omega2.

# Description

Get the latest NodeJS (v16.19.0) package compiled and running for the Omega2 on OpenWRT 22.03. 

If this is not possible, we should understand exactly why.

## Background

In OpenWRT 19.07 and 18.06, the Node package was supported by the Omega2 out of the box because of the MIPS_FPU_EMULATOR support in the kernel. 

We would like to find out:

1. why FPU emulation is no longer a default option
2. if FPU emulation can be enabled
3. if we can get the latest Node running on the Omega2 in OpenWRT 22.

## Output

- In the event of success: modifications to build system or nodejs package makefile to compile Node v16 for Omega2 and package file for NodeJS v16 that works on Omega2
- Otherwise: Document outlining exactly why we cannot run NodeJS v16 on the Omega2 and why FPU emulation is no longer available for MIPS

# Alternatives Considered

N/A

# Additional Resources

Useful resource we found that can serve as a starting place: https://forum.openwrt.org/t/how-to-enable-mips-fpu-emulator-when-compiling/80997/7

# Reference to existing IPK package file?

Node package makefile in OpenWRT's package repo: https://github.com/openwrt/packages/blob/openwrt-22.03/lang/node/Makefile

jempatel, 2023-03-01:
> 
>     1. why FPU emulation is no longer a default option
OpenWrt community has that option disabled by default for a few reasons as found in forum, one of that was to reduce firmware size. 
> 
>     2. if FPU emulation can be enabled
Yes, but it needs changes in a few configs which are being included when running `make menuconfig` or when `.config` is generated. Though, just enabling FPU emulator won't allow NodeJS package selection in `make menuconfig`

1. Allow NodeJS package selection when config is generated.
To allow NodeJS package selection from `make menuconfig`, It need the `CONFIG_HAS_FPU=y` option set, because NodeJS package `Makefile` is changed for dependencies with `@HAS_FPU` now.

We need to add a new default like `default y if TARGET_ramips_mt76x8`. That will enable `HAS_FPU` for the target, which in turn will allow NodeJS package selection and compilation.

2. Enable `KERNEL_MIPS_FP_SUPPORT` support
`KERNEL_MIPS_FP_SUPPORT` is by default enabled for target `pistachio`. More details can be found in the commit https://github.com/openwrt/openwrt/commit/f6d1adbafb9ac3fc0b8048d71fb19f28cf04d7d0.

We need to add a new default like `default y if TARGET_ramips_mt76x8`. This option will compile the kernel with `KERNEL_MIPS_FP_SUPPORT`
> 
>     3. if we can get the latest Node running on the Omega2 in OpenWRT 22.
Yes, successfully compiled NodeJS from upstream tag v22.03.3. Here is the output how what it looks like.

[image]

> 
> 
> ## Output
> 
>     * In the event of success: modifications to build system or nodejs package makefile to compile Node v16 for Omega2 and package file for NodeJS v16 that works on Omega2
> 

We need changes in `Config*.in` files. Its a small change, and the patch looks like this

```patch
diff --git a/config/Config-kernel.in b/config/Config-kernel.in
index ab827d6a71..5141466133 100644
--- a/config/Config-kernel.in
+++ b/config/Config-kernel.in
@@ -44,6 +44,7 @@ config KERNEL_DEBUG_FS
 config KERNEL_MIPS_FP_SUPPORT
        bool
        default y if TARGET_pistachio
+       default y if TARGET_ramips_mt76x8
 
 config KERNEL_ARM_PMU
        bool
diff --git a/target/Config.in b/target/Config.in
index a6b3351a61..652d9151b2 100644
--- a/target/Config.in
+++ b/target/Config.in
@@ -11,6 +11,7 @@ config HAS_SPE_FPU
        bool
 
 config HAS_FPU
+       default y if TARGET_ramips_mt76x8
        bool
 
 config HAS_DT_OVERLAY_SUPPORT
```

# Note
As we are using SDK, with the above changes, the kernel would be compiled with `KERNEL_MIPS_FP_SUPPORT`, so it would not be possible to install `kmod-*.ipk` from upsteam package repos. So we might have to host kernel module packages to make it available for install.

greenbreakfast, 2023-03-01:
Great work @jempatel! It's good to know it's possible to build NodeJS v16 for the Omega2.

To move forward, let's do this NodeJS customization in the OpenWRT build system. 
I'd like to keep our SDK setup to build just our custom packages from this repo. We'll circle back to using the output of the build system in tandem with our SDK and image builder setup later.

I've made a fork of the build system: https://github.com/OnionIoT/openwrt
Let's approach it in two steps:
1. Set up the fork to build openwrt 22.03.3 firmware and base packages - more info at https://github.com/OnionIoT/openwrt/issues/1
2. Make modifications required to build NodeJS - more info at https://github.com/OnionIoT/openwrt/issues/2

Please take on these issues. Let me know if you hit any roadblocks.