OnionIoT/OpenWRT-Packages #52: P44-Ledchain

DocHardinger, 2023-03-31:
# Context

OpenWrt 22.03.
http://repo.onioniot.com/omega2/images/openwrt-22.03/

# Motivation

Possibility of controlling WS Leds would be a nice feature. As in 18.06. p44-ledchain is working.

# Description

Add a patch for p44-ledchain? to use it on OpenWrt 22.03.

# Alternatives Considered

Using other methods to control WS Leds over hardware pwm.

# Reference to existing IPK package file?

https://github.com/plan44/plan44-feed/tree/master/p44-ledchain

greenbreakfast, 2023-03-31:
Loving the double feedback 👍 

Similar to #51, this depends on customization of the kernel. 
We'll add this in once we're done with the infrastructure setup in https://github.com/OnionIoT/openwrt-imagebuilder-wrapper/issues/4

greenbreakfast, 2023-09-28:
@jempatel please tackle this next. 
Let me know if you think this would be better as a patch to openwrt-imagebuilder-wrapper or if we can do it with another method

jempatel, 2023-09-29:
@greenbreakfast This does not need a patch as it is a kernel package, It can be used as is. But it needs rebuild of all packages and kernel modules. Due to new kernel module, it would generate a new hash.

greenbreakfast, 2023-10-02:
* [ ] Patch added to openwrt-buildsystem-wrapper in [Release 20231002-00](https://github.com/OnionIoT/openwrt-buildsystem-wrapper/releases/tag/20231002-00)
* [ ] build new beta firmware that's based on the newly compiled kernel hash

greenbreakfast, 2023-10-10:
Thanks for the confirmation @TT-E1NFACH 

Added brief documentation to the new beta firmware documentation site: https://documentation.onioniot.com/hardware-interfaces/ledchain

More news about this soon :)

closed 2023-10-10 by greenbreakfast