OnionIoT/OpenWRT-Packages #53: Possibility of using One Wire again

DocHardinger, 2023-04-04:
# Context

OpenWrt 22.03.

# Motivation

Would be nice to use One Wire again in OpenWrt 22.03.

# Description

In 18.06. One Wire is working. But in newer version of OpenWrt w1-gpio-custom has been removed. So would be great to find another solutin to use One Wire again.

# Alternatives Considered

Find a way to use One Wire again

greenbreakfast, 2023-04-06:
We'll look into this, thanks for the suggestion!

greenbreakfast, 2023-10-20:
Hi @jempatel
You mentioned on slack you couldn't get one-wire working in the v0.3.4 firmware. I setup a circuit and was able to read data from a DS18B20 sensor.

I followed the circuit setup from https://docs.onion.io/omega2-starter-kit/starter-kit-temp-sensor.html
And communication communication commands from: https://docs.onion.io/omega2-docs/communicating-with-1w-devices.html

```
root@Omega-F11D:/# ll /sys/devices/w1_bus_master1/w1_master_slave_count
-r--r--r--    1 root     root          4096 Jun 28 20:22 /sys/devices/w1_bus_master1/w1_master_slave_count
root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/w1_master_slave_count
1
root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/w1_master_slaves
28-00000f9ca41e
root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e/w1_slave
9f 01 4b 46 7f ff 01 10 40 : crc=40 YES
9f 01 4b 46 7f ff 01 10 40 t=25937
root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e/w1_slave
e5 01 4b 46 7f ff 0b 10 be : crc=be YES
e5 01 4b 46 7f ff 0b 10 be t=30312
```

Could you push changes to openwrt-22.03 that enable one-wire, and I'll try them out on the same circuit?

jempatel, 2023-10-26:
> Hi @jempatel You mentioned on slack you couldn't get one-wire working in the v0.3.4 firmware. I setup a circuit and was able to read data from a DS18B20 sensor.
> 
> I followed the circuit setup from https://docs.onion.io/omega2-starter-kit/starter-kit-temp-sensor.html And the setup and communication communication commands from: https://docs.onion.io/omega2-docs/communicating-with-1w-devices.html
> 
> ```
> root@Omega-F11D:/# insmod w1-gpio-custom bus0=0,19,0
> [  114.622438] Custom GPIO-based W1 driver version 0.1.1
> root@Omega-F11D:/# [  114.667490] w1_master_driver w1_bus_master1: Attaching one wire slave 28.00000f9ca41e crc d6
> ```
> 
> ```
> root@Omega-F11D:/# ll /sys/devices/w1_bus_master1/w1_master_slave_count
> -r--r--r--    1 root     root          4096 Jun 28 20:22 /sys/devices/w1_bus_master1/w1_master_slave_count
> root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/w1_master_slave_count
> 1
> root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/w1_master_slaves
> 28-00000f9ca41e
> root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e/w1_slave
> 9f 01 4b 46 7f ff 01 10 40 : crc=40 YES
> 9f 01 4b 46 7f ff 01 10 40 t=25937
> root@Omega-F11D:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e/w1_slave
> e5 01 4b 46 7f ff 0b 10 be : crc=be YES
> e5 01 4b 46 7f ff 0b 10 be t=30312
> ```
> 
> Could you push changes to openwrt-22.03 that enable one-wire, and I'll try them out on the same circuit?

Thanks @greenbreakfast , the issue was with the jumper cable at my end, with the new jumper cable, its works fine. I am testing new changes at my end now.

greenbreakfast, 2023-11-07:
@jempatel wanted to check in and see how's it going with one-wire
Is there a roadblock currently? Do you need any help?

jempatel, 2023-11-09:
> @jempatel wanted to check in and see how's it going with one-wire Is there a roadblock currently? Do you need any help?

Hi @greenbreakfast, Its done

I was applying changes to load dtbo without rebooting omega2 after package installation. It's tested and usable now, Changes are pushed into MR: https://github.com/OnionIoT/OpenWRT-Packages/pull/75/files

greenbreakfast, 2023-11-09:
Thanks @jempatel, I tried it out on the beta firmware and it worked great! 

Can you please add some basic documentation to: https://github.com/OnionIoT/documentation/blob/main/docs/hardware-interfaces/one-wire.md

@DocHardinger  Please try it out and let us know how it goes!

---

For reference, here's what I did:

```
opkg update
opkg install onion-dt-overlay-w1-gpio
```

I connected a DS18B20+ sensor to GPIO19 and got this kernel message:

```
[  151.107321] w1_master_driver w1_bus_master1: Attaching one wire slave 28.00000f9ca41e crc d6
```

Then I read the sensor data:

```
root@Omega-f195:/# cat /sys/devices/w1_bus_master1/w1_master_slave_count
1
root@Omega-f195:/# cat /sys/devices/w1_bus_master1/w1_master_slaves
28-00000f9ca41e
root@Omega-f195:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e
cat: read error: Is a directory
root@Omega-f195:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e/
alarms       eeprom       hwmon/       resolution   uevent
conv_time    ext_power    id           subsystem/   w1_slave
driver/      features     name         temperature

root@Omega-f195:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e/w1_slave
80 01 4b 46 7f ff 10 10 c6 : crc=c6 YES
80 01 4b 46 7f ff 10 10 c6 t=24000
root@Omega-f195:/# cat /sys/devices/w1_bus_master1/28-00000f9ca41e/w1_slave
d1 01 4b 46 7f ff 0f 10 6c : crc=6c YES
d1 01 4b 46 7f ff 0f 10 6c t=29062
```

DocHardinger, 2023-11-09:
Great! Will check this out on the weekend

DocHardinger, 2023-11-14:
I get these results in kernel log, but maybe there is something wrong with the hardware:

[ 1268.018639] kmodloader: loading kernel modules from /etc/modules.d/*
[ 1268.063866] Driver for 1-wire Dallas network protocol.
[ 1268.088888] kmodloader: done loading kernel modules from /etc/modules.d/*
[ 1268.242423] kmodloader: loading kernel modules from /etc/modules.d/*
[ 1268.280732] kmodloader: done loading kernel modules from /etc/modules.d/*
[ 1268.388775] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/onewire
[ 1268.399520] gpio-499 (onewire): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[ 1271.210541] w1_master_driver w1_bus_master1: w1_search: max_slave_count 64 reached, will continue next search.
[ 1322.330413] w1_master_driver w1_bus_master1: Attaching one wire slave 00.800000000000 crc 8c
[ 1322.385030] w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
[ 1385.929988] w1_master_driver w1_bus_master1: Attaching one wire slave 00.400000000000 crc 46
[ 1385.986902] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[ 1436.689212] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c00000000000 crc ca
[ 1436.744394] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
[ 1513.128770] w1_master_driver w1_bus_master1: Attaching one wire slave 00.200000000000 crc 23
[ 1513.183644] w1_master_driver w1_bus_master1: Family 0 for 00.200000000000.23 is not registered.
[ 1576.728458] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a00000000000 crc af
[ 1576.784910] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
[ 1614.648265] w1_master_driver w1_bus_master1: Attaching one wire slave 00.600000000000 crc 65
[ 1614.702914] w1_master_driver w1_bus_master1: Family 0 for 00.600000000000.65 is not registered.
[ 1655.128157] w1_master_driver w1_bus_master1: Attaching one wire slave 00.e00000000000 crc e9
[ 1655.183430] w1_master_driver w1_bus_master1: Family 0 for 00.e00000000000.e9 is not registered.
[ 1718.127609] w1_master_driver w1_bus_master1: Attaching one wire slave 00.100000000000 crc 9d
[ 1718.181770] w1_master_driver w1_bus_master1: Family 0 for 00.100000000000.9d is not registered.
[ 1757.247532] w1_master_driver w1_bus_master1: Attaching one wire slave 00.900000000000 crc 11
[ 1757.301931] w1_master_driver w1_bus_master1: Family 0 for 00.900000000000.11 is not registered.
[ 1832.487417] w1_master_driver w1_bus_master1: Attaching one wire slave 00.500000000000 crc db
[ 1832.546157] w1_master_driver w1_bus_master1: Family 0 for 00.500000000000.db is not registered.
[ 1884.447198] w1_master_driver w1_bus_master1: Attaching one wire slave 00.d00000000000 crc 57
[ 1884.501606] w1_master_driver w1_bus_master1: Family 0 for 00.d00000000000.57 is not registered.

greenbreakfast, 2023-11-17:
Yep, we can make that change. What GPIO would you prefer?

I would avoid GPIO11 since the bootloader will always program it to output high (https://docs.onion.io/omega2-docs/using-gpios.html#reset-led-pins)

DocHardinger, 2023-11-19:
What about GPIO41?

DocHardinger, 2023-11-21:
Sounds good!

DocHardinger, 2023-11-22:
GPIO 2 is perfect for us!

greenbreakfast, 2023-12-06:
Closing this issue - I've made a new issue (#76) to track changing the GPIO for the one-wire bus master.

closed 2023-12-06 by greenbreakfast