OnionIoT/OpenWRT-Packages #56: Upgrade to OpenWRT 22.03.5

greenbreakfast, 2023-05-18:
# Context

openwrt-22.03 branch

# Motivation

Now that we have a repeatable process for building customized OpenWRT 22.03 firmware, let’s upgrade the system to build the very latest 22.03.5 release.

# Description

Update the https://github.com/OnionIoT/openwrt-sdk-wrapper and https://github.com/OnionIoT/openwrt-imagebuilder-wrapper repos to use the SDK and Image builder from Onion’s OpenWRT 22.03.5 release.
Includes any changes to the files in this repo.

### What does the output of this improvement look like?

Onion customized firmware based on OpenWRT 22.03.5 available at http://repo.onioniot.com/omega2/images/openwrt-22.03/

# Alternatives Considered

N/A

# Additional Resources

This work depends on https://github.com/OnionIoT/openwrt-buildsystem-wrapper being successfully updated to OpenWRT 22.03.5: https://github.com/OnionIoT/openwrt-buildsystem-wrapper/issues/14

# Reference to existing IPK package file?

OpenWRT 22.03.5 runs on other OpenWRT hardware

greenbreakfast, 2023-07-06:
Thanks @jempatel, SDK and image builder wrapper PRs merged.

Last item is to change the omega2-base package version and release numbers.

jempatel, 2023-07-25:
> New 22.03.5 firmware now available here: http://repo.onioniot.com.s3.amazonaws.com/omega2/images/openwrt-22.03/onion_omega2p-22.03.5-20230713.bin
> 
> It works fine but has an issue with opkg. @jempatel can you please look into the opkg issue, more info below?
> 
> When `opkg update` is run, the `http://downloads.onioniot.com` repos have the correct path, but the `downloads.openwrt.org` repos still point to the 22.03.3 release:
> 
> ```
> root@Omega-f195:/# opkg update
> Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/Packages.gz
> Updated list of available packages in /var/opkg-lists/onion_core
> Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/Packages.sig
> Signature check passed.
> ...
> Downloading http://repo.onioniot.com/omega2/packages/openwrt-22.03.5/onion/Packages.gz
> Updated list of available packages in /var/opkg-lists/onion_openwrt_packages
> Downloading http://repo.onioniot.com/omega2/packages/openwrt-22.03.5/onion/Packages.sig
> Signature check passed.
> # issue here:
> Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/base/Packages.gz
> Updated list of available packages in /var/opkg-lists/openwrt_base
> Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/base/Packages.sig
> Signature check passed.
> ...
> Signature check passed.
> root@Omega-f195:/#
> ```
> 
> Where do we do the repo reorganization again?

we only set version for onion-specific packages, other repos that come from `repositories.conf` comes from image-builder.

greenbreakfast, 2023-07-26:
Thanks @jempatel, PR https://github.com/OnionIoT/openwrt-buildsystem-wrapper/pull/19 resolves the problem with the opkg feeds.

Fix is reflected in the `22.03.5-20230726` firmware

---

feed configuration
```
root@Omega-f195:/# cat /etc/opkg/distfeeds.conf
src/gz openwrt_base https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/base
src/gz openwrt_luci https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/luci
src/gz openwrt_packages https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/packages
src/gz openwrt_routing https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/routing
src/gz openwrt_telephony https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/telephony

root@Omega-f195:/# cat /etc/opkg/customfeeds.conf
src/gz onion_core http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages
src/gz onion_base http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/base
src/gz onion_packages http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/packages
src/gz onion_routing http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/routing
src/gz onion_telephony http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/telephony
src/gz onion_openwrt_packages http://repo.onioniot.com/omega2/packages/openwrt-22.03.5/onion
```

opkg update:
```
root@Omega-f195:/# opkg update
Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/onion_core
Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/Packages.sig
Signature check passed.
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/onion_base
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/onion_packages
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/onion_routing
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/onion_telephony
Downloading http://downloads.onioniot.com/releases/22.03.5/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading http://repo.onioniot.com/omega2/packages/openwrt-22.03.5/onion/Packages.gz
Updated list of available packages in /var/opkg-lists/onion_openwrt_packages
Downloading http://repo.onioniot.com/omega2/packages/openwrt-22.03.5/onion/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
```

closed 2023-07-26 by greenbreakfast