OnionIoT/OpenWRT-Packages #58: Using SPI in Python

TT-E1NFACH, 2023-06-05:
# Context

22.03.3

# Motivation

On older version of Onion firmware it was possible to use spidev in Python (see https://github.com/OnionIoT/python-spidev#installation-on-omega2) but in 22.03. this isnt possible anymore. Couldn't be installed over opkg oder pip. 

# Description

It would be nice to use SPI in Python again. For this maybe its possible to get the old version (see https://github.com/OnionIoT/python-spidev#installation-on-omega2)  of spidev running on 22.03.

# Alternatives Considered

Find another way to use SPI in Python or maybe over the console so we can use subprocess.run in Python

# Additional Resources

https://github.com/OnionIoT/python-spidev

greenbreakfast, 2023-09-20:
Implementation: 
Port the `python3-spidev` package to openwrt-22.03 branch

greenbreakfast, 2023-09-25:
@TT-E1NFACH the `python3-spidev` package is now available for the beta openwrt-22.03 firmware
See the updated documentation here: https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md#python-spidev-module

Try it out and let me know how it goes!

greenbreakfast, 2023-09-25:
Small update:
* Updated this repo to change package directory name to `python3-spidev` - now package and directory are named the same 
* Updated https://github.com/OnionIoT/openwrt-sdk-wrapper to actually build the `python3-spidev` package
* The `python3-spidev` package is now being built and is part of the onion package repo
  [image]

Upshot: 
The `python3-spidev` package can now be installed on any Omega2 running the [beta openwrt-22.03 firmware](https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md#installing-the-new-firmware)

TT-E1NFACH, 2023-09-28:
It seems to work for us with new 22.03.5 Firmware, anyway we got an electrical issue on our site but seems that the MCP3008 is reachable and can communicate over spi.

greenbreakfast, 2023-09-28:
@TT-E1NFACH glad to hear you've gotten spi communication up and running. Hope the electrical issue gets resolved soon.

I'm going to go ahead and this issue, feel free to open it if you run into any issues

closed 2023-09-28 by greenbreakfast