OnionIoT/OpenWRT-Packages #59: Kernel obsolete

wasantosfi, 2023-06-06:
# Context
Hardware: Omega2+ with Expansion Dock 
Version: 0.3.3 b256

# Motivation
Need to follow the following procedure (https://openwrt.org/docs/guide-user/virtualization/docker_host#install_docker_community_edition) but isn't possible even to start, once the kernel is very obsolete

# Description
Tried to execute the following command:
opkg install kmod-veth uxc ujail-console

# Alternatives Considered
None.

# Additional Resources
Output:

 opkg install kmod-veth uxc ujail-console
Installing kmod-veth (4.14.206-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06-SNAPSHOT/targets/ramips/mt76x8/packages/kmod-veth_4.14.206-1_mipsel_24kc.ipk
Unknown package 'uxc'.
Unknown package 'ujail-console'.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-veth:
 *     kernel (= 4.14.206-1-da4f999cff891c50c6d11774cc24f4e2) 
 * opkg_install_cmd: Cannot install package kmod-veth.
 * opkg_install_cmd: Cannot install package uxc.
 * opkg_install_cmd: Cannot install package ujail-console.

# Reference to existing IPK package file?
Has this been done for other OpenWRT hardware?
I don't think so.

greenbreakfast, 2023-06-13:
@wasantosfi Thanks for submitting an issue. Using an Omega2 to run a docker container is interesting, do you mind sharing a little bit more about your use case?

On the practical side, yes, the stable v0.3.3 firmware uses the older 4.14 kernel. 

I suggest trying out our new beta firmware - it's based on OpenWRT 22.03.2 and runs Linux Kernel 5.10 - so it should suit your needs.

You can follow these intructions on installing the beta firmware: https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md#installing-the-new-firmware

Please see the rest of that [technical document](https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md) for more info on using the firmware. 

We've taken a new approach with this firmware and started with a blank slate. So it doesn't have all of the utilities and packages that are included in the existing stable (v0.3.3) Omega2 firmware.

If there's something missing from the firmware that you feel is essential, please make a new 
Improvement Suggestion/Feature Request issue in this repo.

*There's more info about the beta firmware and our initiative around it on our forum: https://community.onion.io/topic/4963/beta-openwrt-22-03-firmware-available-your-feedback-is-needed*

Good luck, let us know how it goes!

wasantosfi, 2023-06-14:
Hi @greenbreakfast , thank you for your procedure, I could successfully upgrade my board to **version 22.03.3 build 20230526**, it can connect to internet as well.
Just an update in the git documentation, where we have the "_wget http://repo.onioniot.com/omega2/images/openwrt-22.03/onion_omega2p-22.03.2-20230221.bin_" command for example, it didn't work here, but referencing the aws hosted repo as "_wget http://repo.onioniot.com.s3.amazonaws.com/omega2/images/openwrt-22.03/onion_omega2p-22.03.3-20230526.bin_" command worked.

I tried to install the kmod-veth uxc procd-ujail procd-ujail-console packages but it was unsuccessfull due to openwrt repo not containing 3 of 4 of them (at least not the built in repos).

My intention was to use as docker host, by using a RTOS (real-time operating system) container but maybe it can be way above the Omega2+ hardware resource can handle. Anyway, no problem at all, I can change focus to another activities.

Should I open a new feature request in order to build a new OnionOS variant focused on real-time (preemptive) operations? If possible.

greenbreakfast, 2023-06-16:
@wasantosfi thanks for the heads up about the wget command - both worked for me but I've updated the documentation just in case.

> If you need, I can do some tests using this version and build.

Definitely! Please try using this latest version and let us know where the experience is lacking or what functionality is missing. Or where our [technical document](https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md) could be improved.

> Should I open a new feature request in order to build a new OnionOS variant focused on real-time (preemptive) operations? If possible.

Yes, please open a new issue so we can better track this. I can't guarantee we'll take it up in the near or mid term, but it will be helpful to learn more about the use cases where an RTOS is required.

greenbreakfast, 2023-07-17:
Closing this issue - feel free to open a new issue about RTOS when you're ready

closed 2023-07-17 by greenbreakfast