OnionIoT/OpenWRT-Packages #61: Using hardware pwm the easy way (Bring back onion tool)

TT-E1NFACH, 2023-06-29:
# Context

OpenWrt 22.03

# Motivation

In old version of your Onion Firmware (0.3.3) it was possible to use `onion pwm <CHANNEL> <DUTY CYCLE> <FREQUENCY>`. This was an easy way to use PWM.

# Description

Bring the old onion tool back to OpenWrt 22.03 to make use of PWM and other stuff easier.

# Alternatives Considered

Writing a python module which uses the console commands to make use of pwm easier

# Additional Resources

https://docs.onion.io/omega2-docs/generating-pwm-signals.html

greenbreakfast, 2023-06-29:
@TT-E1NFACH agreed, the `onion pwm` command is much friendlier.

Which other parts of the `onion` utility do you find useful?

TT-E1NFACH, 2023-07-01:
If at all only the timeusage function would be another useful function for us.

greenbreakfast, 2023-08-09:
@TT-E1NFACH The onion script is now available for the beta firmware! 
It's part of a new `omega2-script` package that's not included in the firmware but can be installed. For now it only supports  the `pwm` and `time` commands.

Check out the updated documentation here: https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md#enabling-pwm-pins

I'm gonna close this issue but feel free to reopen it if you have any issues.

closed 2023-08-09 by greenbreakfast