OnionIoT/OpenWRT-Packages #65: valid auth.txt not functioning for usb autorun on latest 22.03 firmware build

mdlss, 2023-08-18:
# What package does this bug affect?

omega2-usb-autorun

# **Describe the bug**

ubus call session login appears to fail while valid credentials are present in auth.txt file

# **To Reproduce**

Steps to reproduce the behavior:
1. attempt to use usb autorun on a fresh install with omega2-usb-autorun installed

# **Expected behavior**
the autorun.sh script should be called and executed

# **Version Info (please complete the following information):**
 - Firmware version:22.03.5 20230809
 - Package name:omega2-usb-autorun
 - Package Version:latest
 - Repo branch:openwrt-22.03

greenbreakfast, 2023-08-21:
@jempatel please take a look at this bug when you have a chacne

jempatel, 2023-08-22:
> @jempatel please take a look at this bug when you have a chacne

Sure.

greenbreakfast, 2023-09-08:
@mdlss we've added a fix for this in the latest beta firmware build.

Please download and install the latest firmware `22.03.5-20230908` from http://repo.onioniot.com/omega2/images/openwrt-22.03/, 
then install the `omega2-usb-autorun` package, and trying again.

The issue should be resolved. Let me know how it goes!

greenbreakfast, 2023-10-03:
@mdlss closing this issue, feel free to reopen it if you have any issues

closed 2023-10-03 by greenbreakfast