OnionIoT/OpenWRT-Packages #70: Add OpenWrt 22.03.xx support to the Omega2LTE

EpicLPer, 2023-09-26:
# Context

openwrt-22.03

# Motivation

It would be nice to have an up-to-date version of OpenWrt on the LTE model to use things like OpenVPN "out-of-the-box" without having to search for specific old versions, as example the OpenVPN SSL package would require some tinkering to get working.

# Description

OpenWrt 22.03 would be a nice addition to have specifically for the LTE model.

# Alternatives Considered

Flashing stock OpenWrt directly on the Omega2S+, however there is no image available for it yet.

# Additional Resources

-

# Reference to existing IPK package file?

-

greenbreakfast, 2023-10-10:
@EpicLPer ok, an package for the Omega2 LTE is now available for the OpenWRT 22.03-based firmware. I don't have an Omega2 LTE on hand to try it out - could you try it out? It is beta firmware, but it's stable for everyday use and you'll be able to go back to the v0.3.4 firmware anytime. Steps below:

* Flash latest beta firmware to your device - see https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md#installing-the-new-firmware
* connect to wifi - see https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/technical-doc.md#sta
* install the omega2 lte package
  ```
  opkg update
  opkg install omega2-lte-base
  ```
* follow the instructions from https://onion.io/omega2-lte-guide/#getstarted-cellular to setup the cellular data connection

Let us know how it goes!

EpicLPer, 2023-10-10:
Awesome! Will test this tomorrow and report back then :)

EpicLPer, 2023-10-11:
> @EpicLPer ok, an package for the Omega2 LTE is now available for the OpenWRT 22.03-based firmware. I don't have an Omega2 LTE on hand to try it out - could you try it out? It is beta firmware, but [...]

Heya,

I don't see a specific firmware for the Omega2LTE, the very latest one I can find from yesterday is in this folder for either the Omega2 or Omega2P: http://repo.onioniot.com/omega2/images/openwrt-22.03/

Which one do I have to choose?

EpicLPer, 2023-10-11:
Gave this a try now, I did notice an error message when setting up my APN:

```
root@Omega-XXXX:/etc/opkg# o2lte apn a1.net
> Setting APN to a1.net
> Configuring network settings
[  774.495883] apcli0: deauthenticating from XX:XX:XX:XX:XX:XX by local choice (Reason: 3=DEAUTH_LEAVING)
> Configuring GPS settings
uci: Entry not found
uci: Entry not found
/usr/bin/o2lte: line 134: /etc/init.d/ugps: not found
> Done
root@Omega-XXXX:/etc/opkg# o2lte data enable
> Enabling LTE data connection
> Done
```

Notice the `/usr/bin/o2lte: line 134: /etc/init.d/ugps: not found` line. I guess this is a package for the GPS component of the Omega2LTE which isn't yet installed, so it shouldn't matter too much. Tho for a stable release definitely important :)

greenbreakfast, 2023-10-11:
@EpicLPer yep, for now Omega2 LTE support is in the form of a package and not a separate firmware. Sorry, should have highlighted that.
Yes, we have yet to port the ugps package. We may consider finding an alternative utility - less to maintain.

Were you able to confirm that you're connected to the cellular network and the `wwan0` interface gets an IP address?

EpicLPer, 2023-10-11:
@greenbreakfast Yeah so far everything seemed to work fine, updated/downloaded the packages and all :) Even the ugps package installed and I was able to enable it, however I couldn't test it as I don't have a GPS antenna at hand I could plug in.

greenbreakfast, 2023-10-16:
@EpicLPer how did you end up installing the ugps package?

EpicLPer, 2023-10-17:
> @EpicLPer how did you end up installing the ugps package?

I simply used "opkg install ugps" and it worked, then used the GPS enable command and it didn't complain about missing binaries anymore.

greenbreakfast, 2023-10-19:
@EpicLPer Great to hear! Added brief documentation about this here: https://documentation.onioniot.com/hardware-interfaces/omega2-lte

I'm going to go ahead and close this issue as the Omega2 LTE is now compatible with the OpenWRT 22.03.5 firmware. 
If there's more development/features that would be helpful to you for the Omega2 LTE, let us know by creating new issues!

closed 2023-10-19 by greenbreakfast