OnionIoT/OpenWRT-Packages #76: Change selected GPIO for one-wire DT overlay package

greenbreakfast, 2023-12-06:
# Context
*Specify which branch of this repo or firmware version you're referring to*

`openwrt-22.03` branch
`onion-dt-overlay-w1-gpio` package

# Motivation

*Why are we doing this? What use cases does it support? What problems does it solve? What is the expected outcome?*

Support for one-wire devices was added in #53 with GPIO19 arbitrarily chosen as the one-wire bus master pin

*If your feature request is related to a problem, provide a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]* 

Changing the one-wire bus master pin to a different GPIO would free up GPIO19 (and hardware PWM channel 1).

The new GPIO for the one-wire bus master should ideally be:
* a GPIO that is not multiplexed with other functionality
* pinned out both the Omega2 and Omega2S variants

Good candidates are GPIOs 0, 1, 2, or 3

# Description

*A clear and concise description of what you want to happen.
It should be reasonably clear how the improvement would be implemented.*

Update the `onion-dt-overlay-w1-gpio` package to use GPIO2 

GPIO2 was selected from the candidate list by collaborating with @DocHardinger in issue #53 

*What does the output of this improvement look like?*

A new version of the `onion-dt-overlay-w1-gpio` package released, documentation updated

# Alternatives Considered

*A clear and concise description of any alternative solutions or features you've considered.*

* Keep using GPIO19 - not ideal because it takes up hw PWM channel1
* GPIO11 - avoid since the bootloader will always program it to output high (https://docs.onion.io/omega2-docs/using-gpios.html#reset-led-pins)
* GPIO41 - avoid since not pinned out on the Omega2/2+ 

# Additional Resources

*Add any other context or screenshots about the feature request here.
Include links to any relevant source code repos*

See issue #53 for original implementation of one wire dt overlay package

# Reference to existing IPK package file?

*Has this been done for other OpenWRT hardware?*

N/A

jempatel, 2023-12-18:
@greenbreakfast So we are going to use GPIO2 for one-wire now instead of GPIO19. 

Pls correct me If I am wrong.

greenbreakfast, 2023-12-18:
@jempatel yep, let's use GPIO2 for one-wire

Sorry if that was unclear in the issue. How can I adjust it to make that more clear?

jempatel, 2023-12-19:
@greenbreakfast, It was clear from your message, I dropped a message just to double-confirm before pushing changes because in old firmware it was already using GPIO19

greenbreakfast, 2024-01-22:
Tested on my end and works fine. Will go ahead and close this issue.

@amsiotdeveloper can you please make a new ticket to update the one wire documentation?
The main update is that we now use GPIO2 as the one-wire master pin.

Along with that, here are some additional edits we should make:
- `Enable One-Wire Device Support` section: need to add a step before step 1 to do opkg update
- `Find Connected One-Wire Devices` section, Step 4: need to add space in command after cat

amsiotdeveloper, 2024-01-24:
@greenbreakfast  ticket has been created to update 1-wire article and it has been assigned to Muze 

[https://github.com/OnionIoT/documentation/issues/36](url)

greenbreakfast, 2024-01-24:
Thanks @amsiotdeveloper 
I'm good to close this ticket since the software part is complete