OnionIoT/OpenWRT-Packages #77: Change selected GPIOs for software SPI DT overlay package

greenbreakfast, 2023-12-06:
# Context
*Specify which branch of this repo or firmware version you're referring to*

`openwrt-22.03` branch
`onion-dt-overlay-sw-spi` package

# Motivation

*Why are we doing this? What use cases does it support? What problems does it solve? What is the expected outcome?*

Support for one-wire devices was added in #62 with the software SPI bus GPIOs chosen arbitrarily.

*If your feature request is related to a problem, provide a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]* 

Current pin mapping is problematic:

| SPI Signal | GPIO | Issue                                                                                                                           |
|:----------:|------|---------------------------------------------------------------------------------------------------------------------------------|
| SCK        | 11   | bootloader will always program it to output high by default (https://docs.onion.io/omega2-docs/using-gpios.html#reset-led-pins) |
| MOSI       | 5    | default: I2C bus SDA signal                                                                                                     |
| MISO       | 4    | default: I2C bus SCL signal                                                                                                     |
| CS         | 23   | Not pinned out on Omega2/2+                                                                                                     |

The new GPIOs for the software SPI bus should ideally be:
* not multiplexed with other functionality
* pinned out both the Omega2 and Omega2S variants

# Description

*A clear and concise description of what you want to happen.
It should be reasonably clear how the improvement would be implemented.*

Update the `onion-dt-overlay-sw-spi` package to use the following pin mapping:

| SPI Signal | New GPIO |
|:----------:|----------|
| SCK        | 14       |
| MOSI       | 16       |
| MISO       | 15       |
| CS         | 17       | 

*What does the output of this improvement look like?*

A new version of the `onion-dt-overlay-sw-spi` package released, documentation updated

# Alternatives Considered

*A clear and concise description of any alternative solutions or features you've considered.*

Many other pin mappings are possible. Open to feedback and suggestions.

# Additional Resources

*Add any other context or screenshots about the feature request here.
Include links to any relevant source code repos*

See issue #62 for original implementation of software spi dt overlay package

# Reference to existing IPK package file?

*Has this been done for other OpenWRT hardware?*

N/A

greenbreakfast, 2024-01-22:
@jempatel is there anything specific I need to do to enable the spi bus?
I installed the `onion-dt-overlay-sw-spi` package and rebooted, but I don't see a `/dev/spidev1.0` device

amsiotdeveloper, 2024-02-26:
@jempatel please provide a status update on this ticket and did @greenbreakfast get a reply to the questions above?

greenbreakfast, 2024-04-19:
Closing this issue since the GPIO selection change is complete.
Tracking the issue with no new SPI device file being created in issue #92