OnionIoT/OpenWRT-Packages #78: OpenWrt 23.05

DocHardinger, 2023-12-09:
# Context
Build custom firmware with OpenWrt 23.05

# Motivation
Some packages are old in 22.03 so we need the new OpenWrt version
Escpecially modemmanager package, which seems version 1.20 in 23.05. which is what we need.

# Description

Tried to use https://github.com/OnionIoT/openwrt-imagebuilder-wrapper with command `bash onion_buildenv setup_imagebuilder` and changed profile to 23.05.02 but there is a problem with base path "http://downloads.onioniot.com.s3.amazonaws.com/releases/", there is no version 23.05. . So i used the original openwrt url "https://downloads.openwrt.org/releases/" but packages are missing when trying to build. Sure its because there are no onion sources for 23.05. 

So please test 23.05 and build the packages

greenbreakfast, 2023-12-12:
Thanks! We'll put this in our backlog

greenbreakfast, 2025-01-06:
Completed in https://github.com/OnionIoT/OpenWRT-Packages/issues/85, https://github.com/OnionIoT/OpenWRT-Packages/issues/86, https://github.com/OnionIoT/openwrt-buildsystem-wrapper/issues/24, and https://github.com/OnionIoT/openwrt-buildsystem-wrapper/issues/25

closed 2025-01-06 by greenbreakfast