OnionIoT/OpenWRT-Packages #80: USB automount does not work properly

mh1412, 2023-12-11:
# What package does this bug affect?

omega2-base

# **Describe the bug**

USB drive does not get mounted when plugged in. 
/mnt/sda1 directory does not get removed when USB drive is unplugged

# **To Reproduce**

Steps to reproduce the behavior:
1. after plugging in USB drive /mnt/sda1 got created but no content is shown
    - `df` command does not show any mount point for USB drive
2. after unplugging USB drive /mnt/sda1 folder still exists, still empty

# **Expected behavior**
When USB drive is plugged in content of the drive should be available under /mnt/sda1
When USB drive is unplugged /mnt/sda1  directory should be removed 

# **Screenshots or Console output**

USB drive plugged in
```
root@Omega-f12d:/mnt# [ 1315.041621] usb 1-1: new high-speed USB device number 7 using ehci-platform
[ 1315.322323] usb-storage 1-1:1.0: USB Mass Storage device detected
[ 1315.343820] scsi host0: usb-storage 1-1:1.0
[ 1316.413481] scsi 0:0:0:0: Direct-Access              USB DISK 3.0     PMAP PQ: 0 ANSI: 6
[ 1317.555909] sd 0:0:0:0: [sda] 30299520 512-byte logical blocks: (15.5 GB/14.4 GiB)
[ 1317.564378] sd 0:0:0:0: [sda] Write Protect is off
[ 1317.569999] sd 0:0:0:0: [sda] No Caching mode page found
[ 1317.575490] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 1317.589026]  sda: sda1
[ 1317.598961] sd 0:0:0:0: [sda] Attached SCSI removable disk

root@Omega-f12d:/mnt# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 4608      4608         0 100% /rom
tmpfs                    61176        76     61100   0% /tmp
tmpfs                    61176        60     61116   0% /tmp/root
tmpfs                      512         0       512   0% /dev
/dev/mtdblock6           25664       748     24916   3% /overlay
overlayfs:/overlay       25664       748     24916   3% /
root@Omega-f12d:/mnt#
root@Omega-f12d:/mnt# ls /mnt/sda1/
root@Omega-f12d:/mnt#
root@Omega-f12d:/mnt#
```
USB drive is unplugged
```
root@Omega-f12d:/mnt# [ 1356.928075] usb 1-1: USB disconnect, device number 7

root@Omega-f12d:/mnt# ls /mnt
sda1
root@Omega-f12d:/mnt#
```

# **Version Info (please complete the following information):**
- Firmware version: [22.03.5 b20231109]
- Package name [omega2-base]
- Package Version [22.03.5-20231109]
- Repo branch (if applicable) [openwrt-22.03]

# **Additional context**
N/A

jempatel, 2024-01-05:
`usbAutorun` script is part of the package `omega2-usb-autorun`.

By default `omega2-usb-autorun` is not part of the firmware.

Can you try the following steps?

1. Unplug USB
2. Install `omega2-usb-autorun`

```sh
opkg update
opkg install omega2-usb-autorun
```

3. Plug USB

amsiotdeveloper, 2024-02-26:
@jempatel I am aware you were away for a while, so please provide a status update on this ticket.

YevheniiOnishchenkoLemberg, 2024-05-22:
@mh1412, there may be an issue with your USB device file system. I formatted my usb (for the test) to use ntfs, ext4, fat32, exfat filesystem types. Automount worked perfectly for ext4 and fat32 but failed for ntfs and exfat. And this is because there was no such kernel module installed for NTFS and exFAT. When I loaded them - I got success. I used onion_omega2p-22.03.3-20230526.bin firmware

greenbreakfast, 2024-06-10:
@amsiotdeveloper can you please make a new documentation task to add [the solution to this mounting problem](https://github.com/OnionIoT/OpenWRT-Packages/issues/80#issuecomment-2124377193) to the documentation