OnionIoT/OpenWRT-Packages #81: No support for Omega2 Pro i2c-exp / adc / python3 / power-dock

t3therdev, 2023-12-13:
# What package does this bug affect?

i2c-exp

# **Describe the bug**

i2c-exp is not able to be compiled on 22.03 due to a lack of support for python2. We have been working on trying to port it over, has there been any work on getting the i2c-exp drivers working for the Omega2 Pro or LTE so we can use i2c to read battery voltages and get power-dock to work?

# **To Reproduce**

# **Expected behavior**
Getting work power-dock, i2c-exp driver for 22.03

# **Screenshots or Console output**

If applicable, add screenshots or terminal output to help explain your problem.

# **Version Info (please complete the following information):**
 - Firmware version: [e.g. 22.03.02 b124]
 - Package name [e.g. omega2-ctrl]
 - Package Version [e.g. 22-1]
 - Repo branch (if applicable) [e.g. openwrt-22.03]

# **Additional context**
Add any other context about the problem here.

greenbreakfast, 2023-12-15:
Thanks for the issue. I'll address both packages here, but in future feel free to make separate issues.

## Power Dock

Reading the battery level is actually very straight-forward. **It involves reading 2 GPIOs and doesn't use the I2C bus at all.**
The battery level is just based on this truth table:
```
	GPIO 18		GPIO16		Battery level
	(level1)		(level0)
	----------------------------------------
	HIGH		LOW 		4/4 - 100%
	HIGH		HIGH		3/4 - 75%
	LOW  		HIGH		2/4 - 50%
	LOW 		LOW		1/4 - 25%
```

Looking back, it was 100% overkill to write a [C program](https://github.com/OnionIoT/power-dock/tree/master) to do this. At this point I would recommend writing a small sh script to read the battery level. See the [beta firmware documentation on GPIOs](https://documentation.onioniot.com/hardware-interfaces/gpio) for more info.

Let me know if you need some pointers with a script

## I2C Exp

Can you give me more context on what you need the i2c-exp packages for?

Our preference is to not port these packages to the new firmware for a few reasons:
- They're an ok I2C implementation, but there's implementations out there that are better in every way (code quality, documentation, maintenance)
- Instead of maintaining old packages that provide marginal benefits, we want to focus our development efforts on what Omega2 users actually find important

That's why I ask about the context. If you need help with getting something specific working, we can work in that direction and/or suggest alternative methods.

greenbreakfast, 2024-12-23:
### Update on I2C Expansion Packages:

We've made the `oled-exp`, `relay-exp`, and `pwm-exp` command line utilities that control the OLED, PWM, and Relay Expansions available as packages for the 23.05 firmware.  

They’re not installed in the firmware by default, you will have to run `opkg update; opkg install oled-exp relay-exp pwm-exp`  to install them. 

Then you can use them on the command line like before. See the documentation for more details:
- https://docs.onion.io/omega2-docs/using-oled-expansion.html
- https://docs.onion.io/omega2-docs/using-relay-expansion.html
- https://docs.onion.io/omega2-docs/using-pwm-expansion.html

We didn’t port the python implementations of these utilities as this would incur a lot of maintenance work. We instead 
 recommend using the subprocess module to run the command line utilities from a python program