OnionIoT/OpenWRT-Packages #86: Test Plan for Part 1 ticket #85

amsiotdeveloper, 2024-02-26:
This is a Subtask for Migration to Major Release Part 1 Ticket #85  

# **Task Owner:**
Who owns this task and will approve the final deliverables? 
Lazar

## **Objectives & Goals:"**
###  1. How does this task relate to the value track(s)?
- All of the effort we put into improving our product lines should provide value to our customers.
- We need to be able to clearly articulate the value each improvement will provide in the following three value tracks:
- This scope is working with the new OpenWRT release so it has a major impact on all 3 value tracks
        1. Development
        2. Deployment
        3. Production
           
###  2. Create a short statement explaining the objectives and goals for this ticket.

- Test basic functionality of compiled firmware to confirm it works, debug as needed
     
# **List of Subtasks:**
- [ ] Part 1: Subtask 1: Test for successful boot process 
- [ ] Part 1: Subtask 2:  Test WiFi AP, WiFi STA
- [ ] Part 1: Subtask 3: Test Ethernet networking
- [ ] Part 1: Subtask 4: Test USB and SD Card functionality
- [ ] Part 1: Subtask 5: Test I2C
- [ ] Part 1: Subtask 6: Test GPIO 

# **Reference Material:**
###  1. Provide hyperlinks for helpful and related information. 
- OnionIoT/OpenWRT-Packages
- OnionIoT/openwrt-sdk-wrapper
- OnionIoT/openwrt-imagebuilder-wrapper

# **Deliverables or Desired Outcome:**
###  1. What are the task owners expectations? 
- Execute the test plan, identify any issues and fix as required. 

###  2. What does the task owner require for a successful completion of the ticket?
    i.e. sample code, drawings, documents etc.
- prepare the firmware for Part 2.
- provide test results, text file or screenshots, demonstrating test results with successful setup/commands  

# **Date Required By:**
###  1. When is the scope required to be completed?
- no specified completion date 

###  2. Is this a hard date?  Specify if the task is a dependency of another task or if it will impact a customer requirement.
- Ticket #85  requires these test results to be completed.
- Part 2 scope requires Part 1 tickets to be completed first.

greenbreakfast, 2024-04-19:
Would like to have an automated and repeatable way for test firmware functionality.
Something along the lines of https://github.com/OnionIoT/omega2-fw-tester - note this is a proof of concept, open to suggestions and other solutions

YevheniiOnishchenkoLemberg, 2024-05-31:
New firmware (http://repo.onioniot.com/omega2-dev/images/openwrt-23.05/) is tested:
[image]

There is a problem using sd card with Speed class 10. It should be fixed in buildsystem repo with a patch.

greenbreakfast, 2024-06-05:
Tested firmware released in #85 with the https://github.com/OnionIoT/omega2-fw-tester

> See release here: https://github.com/OnionIoT/OpenWRT-Packages/releases/tag/23.05.3-20240531
> Firmware available at: http://repo.onioniot.com.s3.amazonaws.com/omega2/images/openwrt-23.05/onion_omega2p-23.05.3-20240531.bin

All tests passing!
[image]

Closing this issue

closed 2024-06-05 by greenbreakfast