OnionIoT/OpenWRT-Packages #91: Build fails if onion-dt-overlay-sw-spi package is included in build

kishangondaliya, 2024-04-16:
# What package does this bug affect?

onion-dt-overlay-sw-spi

# **Describe the bug**

Build fails if onion-dt-overlay-sw-spi package is included (*) in build and works fine if only package building is (M) enabled

LOGS:
cp -fpR /build/build_dir/target-mipsel_24kc_musl/root-ramips /build/build_dir/target-mipsel_24kc_musl/root.orig-ramips
mkdir: cannot create directory '/sys/kernel/config/device-tree': Permission denied
/build/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/onion-dt-overlay-sw-spi.postinst-pkg: line 5: /sys/kernel/config/device-tree/overlays/sw-spi/dtbo: No such file or directory
postinst script ./usr/lib/opkg/info/onion-dt-overlay-sw-spi.postinst has failed with exit code 1
make[2]: *** [package/Makefile:74: package/install] Error 1

# **To Reproduce**

Steps to reproduce the behavior:
Build firmware with onion-dt-overlay-sw-spi package included (*)

# **Expected behavior**
Error-free build

# **Additional context**
If we select the package to be pre-installed, it will create an issue as the postinit script will try to execute in the build system and try to create "/sys" directory, which is not a correct approach.
From https://github.com/OnionIoT/OpenWRT-Packages/issues/62, I can see testing is done by installing the package after booting the base image and not with the entire image build which included this package.

My approach would be to give a warning to a user in the postinit script that a reboot is required after installation if the package is installed after booting else if the package is in the firmware itself, no postinit is required.

YevheniiOnishchenkoLemberg, 2024-05-15:
@kishangondaliya, I've created a [PR](https://github.com/OnionIoT/OpenWRT-Packages/pull/93) with the fix. Will you be able to verify it?

greenbreakfast, 2024-05-17:
@YevheniiOnishchenkoLemberg I merged the PR and tested this out, the build works!

When we test #92 and confirm that works as expected, I will make a release so this fix is available to everyone

---

The test method I followed:
- local clone of [OnionIoT/openwrt-imagebuilder-wrapper](https://github.com/OnionIoT/openwrt-imagebuilder-wrapper)
- modify `profile` to:
  - point to dev package repo `http://repo.onioniot.com/omega2-dev/packages/openwrt-22.03.5/onion` (this is where the [CD system puts development build output](https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/README.md#continuous-deployment-details))
  - include `onion-dt-overlay-sw-spi` package in firmware
- Follow instructions in README to build image

greenbreakfast, 2024-05-21:
The fix for this and #92 was released in https://github.com/OnionIoT/OpenWRT-Packages/releases/tag/22.03.5-20240517

@kishangondaliya I'm closing this issue since we had success on our end. Feel free to reopen if you run into any issues.

PR #93 (2024-05-16): Fix onion-dt-overlay postinst script to have error-free build
Build fails if onion-dt-overlay-sw-spi package is included (*) in build and works fine if only package building is (M) enabled. Onion-dt-overlay package postinst script tries to create /sys/kernel/config/ directory which is not allowed during image building process.
The fix introduces correct usage of IPKG_INSTROOT variable in bash, so the postinst script should be executed only on a target hardware.

Resolves: #91
commit 04933fc0 (2024-05-17): Fix onion-dt-overlay postinst script to have error-free build

Build fails if onion-dt-overlay-sw-spi package is included (*) in
build and works fine if only package building is (M) enabled.
Onion-dt-overlay package postinst script tries to create
/sys/kernel/config/ directory which is not allowed during
image building process.
The fix introduces correct usage of IPKG_INSTROOT variable
in bash, so the postinst script should be executed only
on a target hardware.

Resolves: #91
closed 2024-05-21 by greenbreakfast