OnionIoT/OpenWRT-Packages #92: Software SPI DT Overlay Package doesn't create new SPI device file

greenbreakfast, 2024-04-19:
# What package does this bug affect?

onion-dt-overlay-sw-spi in openwrt-22.03 branch

# **Describe the bug**

After installation of onion-dt-overlay-sw-spi, no new spi device file is created in the Linux OS

# **To Reproduce**

Before installation, only `spidev0.1`, the hardware SPI CS1 bus is available in the `/dev` directory
```
root@Omega-665D:/# ls /dev/
autofs     mtd0       mtd4ro     mtdblock4  shm        ttyS11     ttyS6
bus        mtd0ro     mtd5       mtdblock5  spidev0.1  ttyS12     ttyS7
console    mtd1       mtd5ro     mtdblock6  stderr     ttyS13     ttyS8
full       mtd1ro     mtd6       null       stdin      ttyS14     ttyS9
hwrng      mtd2       mtd6ro     port       stdout     ttyS15     urandom
i2c-0      mtd2ro     mtdblock0  ppp        tty        ttyS2      watchdog
kmsg       mtd3       mtdblock1  ptmx       ttyS0      ttyS3      watchdog0
log        mtd3ro     mtdblock2  pts        ttyS1      ttyS4      zero
mem        mtd4       mtdblock3  random     ttyS10     ttyS5
```

Steps to reproduce the behavior:
1. Connect to internet
2. Install the package 
   ```
   opkg update
   opkg install onion-dt-overlay-sw-spi
   ```
3. reboot device
4. Check if device tree overlay is applied: `cat /sys/kernel/config/device-tree/overlays/sw-spi/status`
   ```
   root@Omega-665D:/# cat /sys/kernel/config/device-tree/overlays/sw-spi/status
   applied
   ```
5. Run `ls /dev` again: 
```
root@Omega-665D:/# ls /dev/
autofs     mtd0       mtd4ro     mtdblock4  shm        ttyS11     ttyS6
bus        mtd0ro     mtd5       mtdblock5  spidev0.1  ttyS12     ttyS7
console    mtd1       mtd5ro     mtdblock6  stderr     ttyS13     ttyS8
full       mtd1ro     mtd6       null       stdin      ttyS14     ttyS9
hwrng      mtd2       mtd6ro     port       stdout     ttyS15     urandom
i2c-0      mtd2ro     mtdblock0  ppp        tty        ttyS2      watchdog
kmsg       mtd3       mtdblock1  ptmx       ttyS0      ttyS3      watchdog0
log        mtd3ro     mtdblock2  pts        ttyS1      ttyS4      zero
mem        mtd4       mtdblock3  random     ttyS10     ttyS5
```

**No new SPI device**

# **Expected behavior**

Expecting to see a new `/dev/spidev1.0` device that corresponds to the new software SPI bus

# **Screenshots or Console output**

[image]

# **Version Info (please complete the following information):**
 - Firmware version: 22.03.5 b20240122
 - Package name onion-dt-overlay-sw-spi
 - Package Version: 1.0-3
 - Repo branch: openwrt-22.03

# **Additional context**

* Software SPI through device tree overlay originally implemented in #62 - tested that device tree overlay was applied but did not test SPI device file operation
* Software SPI GPIOs changed in #77

greenbreakfast, 2024-04-23:
Areas to investigate:
- Package definition doesn't include any kernel spi drivers
  - probably need to also install `kmod-spi-bitbang`, `kmod-spi-dev`, `kmod-spi-gpio` for this to work
  - https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/onion-dt-overlay/Makefile#L82
- DTS fragment 
  - check if the client specification in the dts fragment is compatible with the existing harware spi bus
  - https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-22.03/onion-dt-overlay/src/sw-spi.dts#L21

greenbreakfast, 2024-05-17:
Confirming `/dev/spidev1.0` is created after installing latest version of the onion-dt-overlay-sw-spi package
Will make a new release

---
```
root@Omega-665D:/tmp# opkg update
...
root@Omega-665D:/tmp# wget http://repo.onioniot.com.s3.amazonaws.com/omega2-dev/
packages/openwrt-22.03.5/onion/onion-dt-overlay-sw-spi_1.0-5_mipsel_24kc.ipk
Downloading 'http://repo.onioniot.com.s3.amazonaws.com/omega2-dev/packages/openwrt-22.03.5/onion/oni'
Connecting to 3.5.29.229:80
Writing to 'onion-dt-overlay-sw-spi_1.0-5_mipsel_24kc.ipk'
onion-dt-overlay-sw- 100% |*******************************|  1480   0:00:00 ETA
Download completed (1480 bytes)

root@Omega-665D:/tmp# opkg install onion-dt-overlay-sw-spi_1.0-5_mipsel_24kc.ipk

Upgrading onion-dt-overlay-sw-spi on root from 1.0-3 to 1.0-5...
Installing kmod-lib-crc-itu-t (5.10.176-1) to root...
Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/kmod-lib-ck
Installing kmod-lib-crc7 (5.10.176-1) to root...
Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/kmod-lib-ck
Installing kmod-mmc-spi (5.10.176-1) to root...
Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/kmod-mmc-sk
Installing kmod-spi-bitbang (5.10.176-1) to root...
Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/kmod-spi-bk
Installing kmod-spi-gpio (5.10.176-1) to root...
Downloading http://downloads.onioniot.com/releases/22.03.5/targets/ramips/mt76x8/packages/kmod-spi-gk
Configuring kmod-spi-bitbang.
Configuring kmod-spi-gpio.
[  308.244755] kmodloader: loading kernel modules from /etc/modules.d/*
[  308.290927] spidev@0 enforce active low on chipselect handle
[  308.318023] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring kmod-lib-crc7.
[  308.454281] kmodloader: loading kernel modules from /etc/modules.d/*
[  308.486379] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring kmod-lib-crc-itu-t.
[  308.624791] kmodloader: loading kernel modules from /etc/modules.d/*
[  308.656039] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring kmod-mmc-spi.
[  308.793352] kmodloader: loading kernel modules from /etc/modules.d/*
[  308.826469] kmodloader: done loading kernel modules from /etc/modules.d/*
Configuring onion-dt-overlay-sw-spi.
root@Omega-665D:/tmp# ls /dev/
autofs     mtd0       mtd4ro     mtdblock4  shm        ttyS10     ttyS5
bus        mtd0ro     mtd5       mtdblock5  spidev0.1  ttyS11     ttyS6
console    mtd1       mtd5ro     mtdblock6  spidev1.0  ttyS12     ttyS7
full       mtd1ro     mtd6       null       stderr     ttyS13     ttyS8
hwrng      mtd2       mtd6ro     port       stdin      ttyS14     ttyS9
i2c-0      mtd2ro     mtdblock0  ppp        stdout     ttyS15     urandom
kmsg       mtd3       mtdblock1  ptmx       tty        ttyS2      watchdog
log        mtd3ro     mtdblock2  pts        ttyS0      ttyS3      watchdog0
mem        mtd4       mtdblock3  random     ttyS1      ttyS4      zero
root@Omega-665D:/tmp# ls /dev/spidev1.0
/dev/spidev1.0
```

greenbreakfast, 2024-05-17:
@amsiotdeveloper now that this is fixed, can you please make a new documentation ticket to update the software spi article.

The GPIOs used are now:

| SPI Signal | New GPIO |
|:----------:|----------|
| SCK        | 14       |
| MOSI       | 16       |
| MISO       | 15       |
| CS         | 17       | 

And a reboot is no longer required after installing the package for the new SPI device to be available

amsiotdeveloper, 2024-05-21:
@greenbreakfast ticket 91 on Documentation board has been created and assigned to Tomas
https://github.com/OnionIoT/documentation/issues/91

PR #94 (2024-05-17): Add kernel dependencies for onion-dt-overlay-sw-spi package
Resolves: #92
commit 695732aa (2024-05-17): Add kernel dependencies for onion-dt-overlay-sw-spi package

Resolves: #92
closed 2024-05-21 by greenbreakfast