OnionIoT/OpenWRT-Packages #100: DT Overlay Functionality not available after reboot

greenbreakfast, 2024-06-10:
# What package does this bug affect?

`onion-dt-overlay-sw-spi` and `onion-dt-overlay-w1-gpio` in the `openwrt-23.05` branch

And most likely in `openwrt-22.03` branch as well

# **Describe the bug**

The functionality added by installing Device Tree Overlay packages is only available until the device is powered off or rebooted. 
Once the device reboots, the DT overlay seems to still be in place, but the functionality is not available. That is the additional software SPI bus and the one-wire master are not available.

# **To Reproduce**

*Used the [OnionIoT/omega2-fw-tester](https://github.com/OnionIoT/omega2-fw-tester) to test this quickly*

Steps to reproduce the behavior:
1. Flash the latest [`onion_omega2p-23.05.3-20240610.bin` firmware](http://repo.onioniot.com.s3.amazonaws.com/omega2-dev/images/openwrt-23.05/onion_omega2p-23.05.3-20240610.bin) to the device
2. Setup wifi or otherwise connect the device to the internet
3. On connected computer, run `pytest -v test_device.py --json-config test_cases/400_opkg_update.json test_cases/600_dt_overlay.json`, observe all tests pass
  [image]

4. Reboot the Omega2 device
5. On connected computer, run `pytest -v test_device.py --json-config test_cases/600_dt_overlay.json`, observe the tests that check functionality fail:
  [image]

  

# **Expected behavior**

Expecting Software SPI and One-Wire Master to be available and work properly after a reboot or power off

# **Screenshots or Console output**

After initial installation of the dt overlay packages (step 3 above), tested manually on command line and both SW SPI and One-Wire worked. Could even read temperature data from a connected sensor:
[image]

After rebooting the device (step 4 above), tested manually on command line again. No SW SPI device available, not one-wire master device available:
[image]
 

# **Version Info (please complete the following information):**
 - Firmware version: 23.05.3-20240610
 - Package name: `onion-dt-overlay-sw-spi` and `onion-dt-overlay-w1-gpio`
 - Package Version: `1.0-5`
 - Repo branch (if applicable) `openwrt-23.05` branch

greenbreakfast, 2024-06-12:
Tested on **22.03.5**-20240517 firmware and did not have the same issue. So **this issue is only present on the 23.05.3 firmware!**

After fresh fw install:
[image]

After reboot:
[image]

YevheniiOnishchenkoLemberg, 2024-06-20:
Fixed in PR "Fix dt-overlay patch #35". Results below are for initial boot and after pressing the reset button:

[image]

[image]

greenbreakfast, 2024-07-26:
Confirming this issue is resolved in the [`23.05.3-20240724` release](https://github.com/OnionIoT/OpenWRT-Packages/releases/tag/23.05.3-20240724).

**Successfully** tested on [`23.05.3-20240724` firmware](http://repo.onioniot.com.s3.amazonaws.com/omega2/images/openwrt-23.05/onion_omega2p-23.05.3-20240724.bin) from [`23.05.3-20240724` release](https://github.com/OnionIoT/OpenWRT-Packages/releases/tag/23.05.3-20240724)

Followed [test procedure from original post](https://github.com/OnionIoT/OpenWRT-Packages/issues/100#issue-2344741904)
After fresh fw install, dt-overlay installation and functionality checks pass:
[image]

After reboot, dt-overlay functionality checks pass:
[image]

Also tested one-wire device after reboot, works as expected. Just noticed a few ghost one-wire slave devices, but that can be investigated at a later date:
[image]