OnionIoT/OpenWRT-Packages #104: When in AP+STA mode, The AP doesn't work if the STA can't access the network it was configured for

djmoran64, 2024-11-26:
# What package does this bug affect?
It affects the wifi, I don't know the package

# **Describe the bug**

I am evaluating moving from 0.3.4 to the new 23.5.03 firmware.  The new firmware comes with the AP configured but not the STA like the old one did.  Per the documentation, I un-disabled `wireless.client` and set the ssid and password to my local network.  All worked as expected.  The Omega had an Access Point as well as access to my local network.  When I booted the Omega in a different location having a different wifi network, the STA client could not connect and that kept the AP from starting up as well.  In version 0.3.4, this was not the case.   The AP worked even when the STA couldn't connect.

# **To Reproduce**

Steps to reproduce the behavior:
1. cd /tmp
2. wget http://repo.onioniot.com.s3.amazonaws.com/omega2/images/openwrt-23.05/onion_omega2p-23.05.3-20241024.bin
3. sysupgrade -F -n -v onion_omega2p-23.05.3-20241024.bin
4. Verify AP is working - ifconfig:
```
    ra0:      Link encap:Ethernet  HWaddr 40:A3:6B:C5:64:C2  
                 inet addr:192.168.3.1  Bcast:192.168.3.255  Mask:255.255.255.0
                 inet6 addr: fe80::42a3:6bff:fec5:64c2/64 Scope:Link
                 UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
                 RX packets:15558 errors:0 dropped:0 overruns:0 frame:0
                 TX packets:13629 errors:0 dropped:0 overruns:0 carrier:0
                 collisions:0 txqueuelen:1000 
                 RX bytes:1223652 (1.1 MiB)  TX bytes:1338577 (1.2 MiB)
```
6. uci set wireless.client.disabled='0'
7. uci set wireless.client.ssid='ssid'         <==== Use correct name for SSID
8. uci set wireless.client.key='password'
9. uci commit wireless
10. wifi 
```
      [ 2136.453731] apcli0: authenticate with dc:eb:69:44:44:f8
      [ 2136.677677] apcli0: send auth to dc:eb:69:44:44:f8 (try 1/3)
      [ 2136.706332] apcli0: authenticated
      [ 2136.717374] apcli0: associate with dc:eb:69:44:44:f8 (try 1/3)
      [ 2136.726084] apcli0: RX AssocResp from dc:eb:69:44:44:f8 (capab=0x1531 status=0 aid=10)
      [ 2136.734472] apcli0: associated
      [ 2136.777635] IPv6: ADDRCONF(NETDEV_CHANGE): apcli0: link becomes ready
      [ 2137.367248] IPv6: ADDRCONF(NETDEV_CHANGE): ra0: link becomes ready
```
11. Verify AP is working - ifconfig:
```
      ra0:      Link encap:Ethernet  HWaddr 40:A3:6B:C5:64:C2  
                   inet addr:192.168.3.1  Bcast:192.168.3.255  Mask:255.255.255.0
                   inet6 addr: fe80::42a3:6bff:fec5:64c2/64 Scope:Link
                   inet6 addr: fdc5:e9fa:3d1d:10::1/60 Scope:Global
                   UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
                   RX packets:0 errors:0 dropped:0 overruns:0 frame:0
                   TX packets:13 errors:0 dropped:0 overruns:0 carrier:0
                   collisions:0 txqueuelen:1000 
                   RX bytes:0 (0.0 B)  TX bytes:2048 (2.0 KiB)  
     apcli0    Link encap:Ethernet  HWaddr 42:A3:6B:C5:64:C2
                   inet addr:10.0.0.63  Bcast:10.0.0.255  Mask:255.255.255.0
                   inet6 addr: fe80::40a3:6bff:fec5:64c2/64 Scope:Link
                   UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
                   RX packets:1792 errors:0 dropped:0 overruns:0 frame:0
                   TX packets:13 errors:0 dropped:0 overruns:0 carrier:0
                   collisions:0 txqueuelen:1000 
                   RX bytes:618574 (604.0 KiB)  TX bytes:2322 (2.2 KiB)
```
12. uci set wireless.client.ssid='BADSSID'         <==== Use INCORRECT name for SSID
13. uci set wireless.client.key='BADPASSWORD'
14. uci commit wireless
15. wifi 
16. Verify - The access point wifi is NOT advertised and NOT available.  
17. ifconfig:
```
      apcli0    Link encap:Ethernet  HWaddr 42:A3:6B:C5:64:C2  
                UP BROADCAST MULTICAST  MTU:1500  Metric:1
                RX packets:0 errors:0 dropped:0 overruns:0 frame:0
                TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
                collisions:0 txqueuelen:1000 
                RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
      ra0       Link encap:Ethernet  HWaddr 40:A3:6B:C5:64:C2  
                inet addr:192.168.3.1  Bcast:192.168.3.255  Mask:255.255.255.0
                UP BROADCAST MULTICAST  MTU:1500  Metric:1
                RX packets:0 errors:0 dropped:0 overruns:0 frame:0
                TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
                collisions:0 txqueuelen:1000 
                RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B):
```
17. See attached wifi_log.txt for full system messages, excerpt here:
```
          ...
          Tue Nov 26 15:35:24 2024 daemon.notice hostapd: Restart interface for phy phy0
          Tue Nov 26 15:35:25 2024 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=g supported_rates=60 90 120 180 240 360 480 540 basic_rates=60 120 240 beacon_int=100 noscan=1 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][RX-STBC1] channel=acs_survey  interface=ra0 bssid=40:a3:6b:c5:64:c2 ctrl_interface=/var/run/hostapd bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=12345678 wpa_psk_file=/var/run/hostapd-ra0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=Omega-64C2 wpa_disable_eapol_key_retries=0 wpa_key_mgmt=WPA-PSK okc=0 disable_pmksa_caching=1 dynamic_vlan=0 vlan_naming=1 vlan_no_bridge=1 vlan_file=/var/run/hostapd-ra0.vlan qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56 #default_macaddr nas_identifier=40a36bc564c2
          Tue Nov 26 15:35:25 2024 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
          Tue Nov 26 15:35:25 2024 daemon.notice netifd: Interface 'wlan' is enabled
          Tue Nov 26 15:35:25 2024 daemon.notice netifd: Interface 'wlan' is setting up now
          Tue Nov 26 15:35:25 2024 daemon.notice hostapd: ra0: interface state UNINITIALIZED->ACS
          Tue Nov 26 15:35:25 2024 daemon.notice hostapd: ra0: ACS-STARTED
          Tue Nov 26 15:35:25 2024 daemon.notice netifd: Interface 'wlan' is now up
          Tue Nov 26 15:35:26 2024 daemon.notice wpa_supplicant[1167]: Set new config for phy phy0
          Tue Nov 26 15:35:27 2024 daemon.notice netifd: Wireless device 'radio0' is now up
          Tue Nov 26 15:35:27 2024 user.notice firewall: Reloading firewall due to ifup of wlan (ra0)
          Tue Nov 26 15:35:27 2024 daemon.notice hostapd: ra0: AP-DISABLED
          Tue Nov 26 15:35:27 2024 daemon.err hostapd: hostapd_free_hapd_data: Interface ra0 wasn't started
          Tue Nov 26 15:35:27 2024 daemon.notice hostapd: nl80211: deinit ifname=ra0 disabled_11b_rates=0
          Tue Nov 26 15:35:27 2024 daemon.notice hostapd: ra0: interface state ACS->DISABLED
          Tue Nov 26 15:35:27 2024 daemon.notice netifd: Interface 'wlan' is now down
          Tue Nov 26 15:35:27 2024 daemon.notice netifd: Interface 'wlan' is disabled
          ...
```
I tried rebooting, didn't help
I tried setting ```option disassoc_low_ack '0'``` on the ap
I tried a chatgpt suggestion: ... creating a bridge between the AP (ra0) and a fallback network interface (e.g., lan) to ensure AP availability even when the STA interface fails. Modified the /etc/config/network:
```config interface 'wlan'
    option device 'br-wlan'
    option proto 'static'
    option ipaddr '192.168.3.1'
    option netmask '255.255.255.0'

config device
    option name 'br-wlan'
    option type 'bridge'
    list ports 'ra0'
```
I feel like I googled and looked at the OpwnWRT docs enough to warrant creating this issue :-)

# **Expected behavior**
I expect that the AP will still function when STA can't connect to the local network.

# **Screenshots or Console output**
See attached ```wifi_log.txt``` for the system messages sent after changing sta from a valid working SSID to an invalid network SSID and running the wifi command.

See attached ```wireless``` and ```network``` /etc/config files

[wifi_log.txt](https://github.com/user-attachments/files/17924989/wifi_log.txt)
[network.txt](https://github.com/user-attachments/files/17925013/network.txt)
[wireless.txt](https://github.com/user-attachments/files/17925014/wireless.txt)
 

# **Version Info (please complete the following information):**

 onion_omega2p-23.05.3-20241024.bin

# **Additional context**
I think I said it all above

greenbreakfast, 2024-12-02:
Hi @djmoran64, thanks for the detailed bug report!

We've looked into this and were able to replicate it on our end - and have confirmed this is expected behaviour for the mt76 wifi driver. **It cannot successfully host a WiFi AP if the STA is configured but cannot be found.** 

This is likely due to a design decision in the mt76 driver on how it handles the AP and STA virtual interfaces sharing one physical radio. The wifi warp core driver in the v0.3.4 firmware made other trade-offs (worse nl80211 support) in order to attempt to decouple the two virtual interfaces.

---

For your use case, what would be an acceptable resolution? Better documentation, ideas on how to mitigate this, examples on how to mitigate this? 

Let us know and we'll see what we can do

djmoran64, 2024-12-11:
@greenbreakfast Thanks for the reply and looking into this.  I understand that there are definitely tradeoffs here.  I think the the documentation definitely should be updated as it is a difference for people upgrading.  

I appreciate in any ideas you have to help mitigate this.  My initial thoughts are to add a daemon process to watch out for this situation.  Monitor for the AP to go down `iw dev ra0 info | grep "ssid"` and STA `iw dev apcli0 link | grep "Connected to"` then disable the STA interface 
```
uci set wireless.client.disabled='1'
uci commit wireless
wifi
```
Then every 5 minutes or so scan for the STA ssid to be available with  `iwscan`

For my use case, the omega is sealed in an underwater camera so losing the AP is bad and would force opening up the housing.  So - it is expected that the STA would not be available when the camera is deployed and come back when recovered on the boat.

greenbreakfast, 2024-12-18:
@djmoran64 we've updated the documentation to point this issue out. Take a look here and please let me know if it could be improved: https://documentation.onioniot.com/networking/wifi#ap-not-accessible-if-sta-cannot-connect

---

I agree that running a script run at some interval to check if the STA is connected is a good idea. IMO a generic script would just make sure the AP is accessible. 

Here's how I envision it:

> It would first check:
> 1. if the wireless client interface is enabled
> 2. if the wireless client is not actively connected to a network
> 
> If both of the conditions are true: then it should disable the wireless client interface so the AP is accessible

Cron could be used to run the script at some interval. See the [old docs for more info on using cron](https://docs.onion.io/omega2-docs/running-a-command-on-a-schedule.html)

> How to do the checks:
> - wireless client is enabled if `uci get wireless.client.disabled` is equal 0
> - wireless client not actively connected to a network:
>   - if `iw dev apcli0 link` returns `Not connected`
>   - if `cat /sys/class/net/apcli0/operstate` returns `down`
> 
> How to disable the wireless client interface, save config changes, and restart the wifi radio:
> ```
> uci set wireless.client.disabled='1'
> uci commit
> wifi
> ```

Would it be helpful to have a script that does the above included in the firmware? You would then have a foundation to build on if you needed different behaviour.

greenbreakfast, 2025-01-10:
Hi @djmoran64, happy new year!
Wanted to check in and see if you were able to come to a good resolution for your use case. Let me know if we can do anything to help on our end!