OnionIoT/OpenWRT-Packages #105: Add DTS Overlay example for setting default pin multiplexing

greenbreakfast, 2024-11-27:
# Context

openwrt-23.05 branch

# Motivation

*Why are we doing this? What use cases does it support? What problems does it solve? What is the expected outcome?*

Currently pin multiplexing configuration set by `omega2-ctrl` is not persistent after a reboot. 
Making the pin multiplexing configuration permanent requires making a change to the `/etc/rc.local` boot script. 

DT Overlay is a great way to make this change dynamically in the device tree without having to change the boot script. Addition of packages is a much better way to customize functionality than changing scripts.

This supports any use case that uses the Omega's I/Os in a way that does not match the default configuration.

# Description

*A clear and concise description of what you want to happen.*

Create a DT overlay package that serves as an example for changing the default pin multiplexing for a pin group

*What does the output of this improvement look like?*

Addition of DT overlay package to https://github.com/OnionIoT/OpenWRT-Packages/tree/openwrt-23.05/onion-dt-overlay that changes default pin multiplexing configuration

# Alternatives Considered

*A clear and concise description of any alternative solutions or features you've considered.*

Existing solution is to configure the omega to run the gpio mux command automatically at boot time -> https://documentation.onioniot.com/software/running-command-on-boot

# Additional Resources / Reference to existing IPK package file?

*Has this been done for other OpenWRT hardware?*

See https://github.com/OnionIoT/OpenWRT-Packages/tree/openwrt-23.05/onion-dt-overlay

viteo, 2024-11-28:
you asked about the proposals in [146](https://github.com/OnionIoT/documentation/discussions/146): everything is gpio (except SPI0 😃 )

greenbreakfast, 2024-11-28:
@viteo ok, that makes sense, thanks for the feedback! 
We'll put this into the development queue and I'll post here when something is available to be tested

crispyoz, 2024-11-28:
Set GPIO 18  to PWM would be a good example of how to use overlays default pins

greenbreakfast, 2024-12-17:
@viteo, @crispyoz After working on this, I've come to the conclusion that **using a Device Tree overlay to set the default pin mux setting is not currently possible.** The reason being the DT overlay happens ***after*** pin initialization in the device boot sequence. So the pins are configured to the default pin mux settings before the DT overlay takes place.

Sharing my test notes below for future reference and/or if others want to pick up where I left off.

## Current Recommendations

The [suggestion in the documentation](https://documentation.onioniot.com/hardware-interfaces/pin-multiplexing#making-the-pin-multiplexing-configuration-permanent) to use `/etc/rc.local` to set the pin mux settings at boot to your desired configuration is the best solution for now! 

## Testing Notes

Testing was done using a DT Overlay to set the UART2 pins to GPIO. Able to confirm the device tree on the device was successfully modified, but the pin mux settings did not change.

Here is the DT overlay stub to set the UART2 pins to the GPIO function:
```
/dts-v1/;
/plugin/;

 / {
    fragment@0 {
        target = <&uart2_pins>;
        __overlay__ {
            uart2 {
                groups = "uart2";
                function = "gpio";
            };
        };
    };

    fragment@1 {
        target = <&state_default>;
        __overlay__ {
            uart2 {
                groups = "uart2";
                function = "gpio";
            };
        };
    };
};
```

Created a package following the instructions in the [onion-dt-overlay/README.md file](https://github.com/OnionIoT/OpenWRT-Packages/blob/openwrt-23.05/onion-dt-overlay/README.md)

### Kernel Messages and Overlay Status

After installing the uart2 pin mux dt overlay package and a reboot, kernel messages show the overlay being successfully applied:

```
[    9.359541] mount_root: switching to jffs2 overlay
[    9.367176] overlayfs: upper fs does not support tmpfile.
[    9.380074] urandom-seed: Seeding with /etc/urandom.seed
 - applying device-tree overlay 'pin-mux-set-uart2-to-gpio' -
[    9.537488] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /pinctrl/uart2/uart2/groups
[    9.548126] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /pinctrl/uart2/uart2/function
 - applying device-tree overlay 'sw-spi' -
[    9.600363] procd: - early -
[    9.603741] procd: - watchdog -
[   10.463755] procd: - watchdog -
[   10.468397] procd: - ubus -
```

And the overlay status reads as applied:
```
root@Omega-FBF4:/# cat /sys/kernel/config/device-tree/overlays/pin-mux-set-uart2
-to-gpio/status
applied
```

### Checking Device Tree on Device 
Used the `dtc` utility to read the live device tree on the device
- To install: `opkg update; opkg install dtc` 
- To dump the live device tree: `dtc -I fs -O dts /proc/device-tree > current-device-tree.dts`

Device tree before uart2 pinmux dt overlay package installed: [device-tree-before.dts.txt](https://github.com/user-attachments/files/18171181/device-tree-before.dts.txt)

Device tree after uart2 pinmux dt overlay package installed: [device-tree-after-uart2.dts.txt](https://github.com/user-attachments/files/18171182/device-tree-after-uart2.dts.txt)

Diff:
```
 diff device-tree-before.dts device-tree-after-uart2.dts
570c570
< 				function = "uart2";
---
> 				function = "gpio";
606a607,611
> 			uart2 {
> 				groups = "uart2";
> 				function = "gpio";
> 			};
>
```

-> Changed `pinctrl.uart2.uart2.function` to `gpio` and added `uart2` group to `pinctrl.pinctrl0`

### Checking pin mux settings: ❌

Still show uart2 set to uart function:

```
root@Omega-FBF4:/# omega2-ctrl gpiomux get
Group i2c - [i2c] gpio
Group uart0 - [uart] gpio
Group uart1 - [uart] gpio pwm01
Group uart2 - [uart] gpio pwm23
Group pwm0 - pwm [gpio]
Group pwm1 - pwm [gpio]
Group refclk - refclk [gpio]
Group spi_s - spi_s [gpio] pwm01_uart2
Group spi_cs1 - [spi_cs1] gpio refclk
Group i2s - i2s [gpio] pcm
Group ephy - ephy [gpio]
Group wled - wled [gpio]
```

### My Conclusion

I'm not a kernel expert but judging by the messages in the kernel log:

```
[    0.288640] mt7621_gpio 10000600.gpio: registering 32 gpios in bank 0 with base 0
[    0.296134] mt7621_gpio 10000600.gpio: registering 32 gpios in bank 1 with base 32
[    0.303930] mt7621_gpio 10000600.gpio: registering 32 gpios in bank 2 with base 64
[    0.312038] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.321972] printk: console [ttyS0] disabled
[    0.326159] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 28, base_baud = 2500000) is a 1655A
[    0.334964] printk: console [ttyS0] enabled
[    0.343375] printk: bootconsole [early0] disabled
[    0.353864] 10000d00.uart1: ttyS1 at MMIO 0x10000d00 (irq = 29, base_baud = 2500000) is a 16550A
[    0.363952] 10000e00.uart2: ttyS2 at MMIO 0x10000e00 (irq = 30, base_baud = 2500000) is a 16550A
[    0.375172] spi-mt7621 10000b00.spi: sys_freq: 193333333
[    0.411700] spi-nor spi0.0: w25q256 (32768 Kbytes)
``

It looks like the pins (and the pin mux settings) are configured very early in the boot sequence, less than a second after boot.
While the DT overlays take place about 9 seconds after boot. 

This is likely why the DT overlay to set the UART2 pins to GPIO functionality is successfully applied but does not actually change the pin functionality.

closed 2024-12-17 by greenbreakfast

crispyoz, 2024-12-18:
This looks interesting.  

https://community.nxp.com/t5/i-MX-Processors-Knowledge-Base/How-to-use-linux-device-tree-overlay-from-u-boot-and-apply-it/ta-p/1896858

Currently I am working with owrt 24.10 snapshots, will find time to get my head around this and see if it will work. Current Onion owrt 23 beta is kernel 5.x, but this requires kernel 6.6.3.

greenbreakfast, 2024-12-18:
@crispyoz yep, being able to do dt overlays in the bootloader would resolve this issue. But using rc.local is a valid alternative for most use cases.
We'll keep an eye on this when it's time to move to a newer openwrt release