OnionIoT/OpenWRT-Packages #107: Please add sftp-server to default package list

crispyoz, 2024-12-15:
# Context

Specify which branch of this repo or firmware version you're referring to

23.05.3

# Motivation

Why are we doing this? What use cases does it support? What problems does it solve? What is the expected outcome?

If your feature request is related to a problem, provide a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Needed for upgrading devices using scp

# Description

Please add openssh-sftp-server and openssh-sftp-client  packages to the base files.  Presently if I use the wrapper to create firmware, the packages are not included and then I cannot update the software without manually installed sftp server. I suspect  new players will get tripped up by this as manually install sftp server requires editing of distfeeds.conf.

What does the output of this improvement look like?

# Alternatives Considered

A clear and concise description of any alternative solutions or features you've considered.

# Additional Resources

Add any other context or screenshots about the feature request here.
Include links to any relevant source code repos

# Reference to existing IPK package file?

Has this been done for other OpenWRT hardware?

greenbreakfast, 2024-12-20:
@crispyoz I agree that it's useful to have these packages already installed in the firmware for usability

Added sftp server and client to the firmware image in the [23.05.3-20241220 release](https://github.com/OnionIoT/OpenWRT-Packages/releases/tag/23.05.3-20241220)! 

> I suspect new players will get tripped up by this as manually install sftp server requires editing of distfeeds.conf.

Really? On which firmware version? Users should be able to just do `opkg update; opkg install openssh-sftp-server` and the omega will install it from OpenWRT's package repos

crispyoz, 2024-12-21:
I was playing with the image builder so it was 23.05.3.

closed 2024-12-21 by crispyoz