OnionIoT/OpenWRT-Packages #110: SPI mode 3 (CPOL = 1, CPHA = 1)

denfaj, 2025-01-29:
# What package does this bug affect?

Python spidev

# **Describe the bug**

Hi,
I am having problem on version OpenWRT:
root@Omega-E58B:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="23.05.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.3"
VERSION_ID="23.05.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r23809-234f1a2efa"
OPENWRT_BOARD="ramips/mt76x8"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.3 r238"

I am using python and spidev according to this manual.

My hardware requires SPI mode 3 (CPOL = 1, CPHA = 1) but when I set to spi.mode = 3 and I send bytes using xfer3 like spi.xfer3([0xFF,0xFF,0xFF,0xFF,0xFF,0xFF],0)

I get the following error:
OSError: [Errno 22] Invalid argument

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
File "/root/spi3.py", line 78, in
spi.xfer3([0xFF,0xFF,0xFF,0xFF,0xFF,0xFF],0)
SystemError: <method 'xfer3' of 'SpiDev' objects> returned a result with an exception set

When I look into the log:
root@Omega-E58B:~# dmesg | grep spi
[ 0.375176] spi-mt7621 10000b00.spi: sys_freq: 193333333
[ 0.411913] spi-nor spi0.0: mx25l25635e (32768 Kbytes)
[ 0.417374] 4 fixed-partitions partitions found on MTD device spi0.0
[ 0.423870] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[ 0.431376] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[ 0.439388] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[ 0.446871] OF: Bad cell count for /palmbus@10000000/spi@b00/flash@0/partitions
[ 0.454607] Creating 4 MTD partitions on "spi0.0":
[ 884.823823] spidev spi0.1: setup: unsupported mode bits 3
[ 2315.442212] spidev spi0.1: setup: unsupported mode bits 3
[ 2404.149346] spidev spi0.1: setup: unsupported mode bits 3
[ 2423.349365] spidev spi0.1: setup: unsupported mode bits 3
[ 2453.193319] spidev spi0.1: setup: unsupported mode bits 3
[ 2461.649283] spidev spi0.1: setup: unsupported mode bits 3

I see that spi mode 3 is not supported.

If I set mode to 0 then for instance a simple read operation works only for the first time
response = spi.xfer3([0x50],3)
config_value = (response[0] << 16) | (response[1] << 8) | response[2]
print(f"{config_value:024b}")
First time I get correct response
000000000000000100010111
and second time I get only 0.
000000000000000000000000

Please explain me how to enable mode 3 or how to use Onion with device that requires SPI mode 3.

# **Version Info (please complete the following information):**
NAME="OpenWrt"
VERSION="23.05.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.3"
VERSION_ID="23.05.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r23809-234f1a2efa"
OPENWRT_BOARD="ramips/mt76x8"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.3 r238"

greenbreakfast, 2025-01-31:
@denfaj Thanks for making a github issue to report this bug!

Just a guess but it's possible SPI mode 3 is not supported on the Omega2's hardware SPI bus because the internal flash storage is also connected to this same bus, just on CS0. Perhaps switching between the different modes so quickly may not be supported by the hardware.

But let's try to first verify if this is an issue with the spidev python module. Using the [Omega2 beta firmware](https://documentation.onioniot.com/firmware/installing-firmware), please try [setting up a software SPI bus](https://documentation.onioniot.com/device-tree-overlay/software-spi) and using your same program on the new sw SPI bus. 

If there are issues, this points to an issue in the spidev python module.
If it works with no problems, this suggests there is an incompatibility in the hardware SPI bus.

Let us know how it goes.

denfaj, 2025-02-01:
@greenbreakfast thank your for suggestion. I did set up software SPI bus but since I am using OM-D-EXP together with Omega 2+ I do not have required GPIO's. In manual https://documentation.onioniot.com/device-tree-overlay/software-spi/ it says I have to connect SCK to GPIO 14, MOSI to GPIO 16, MISO to GPIO 15 and CS to GPIO 17. Looking at the pinout I have on module GPIO 15-17 but I am still missing GPIO 14. I see this GPIO is avaliable on Omega 2S,

If I connect according to the pinout https://documentation.onioniot.com/hardware-interfaces/pinouts I do not get proper answer.

This is my code:
```
import sys
import spidev
import time

spi = spidev.SpiDev()
spi.open(0,1)

spi.max_speed_hz=1000000
#spi.mode=3

# AD7190 Commands
RESET_COMMAND = [0xFF] * 5  # Send 40 bits of 1 to reset
COMM_READ = 0x40            # Read operation (MSB = 1)
COMM_WRITE = 0x00           # Write operation (MSB = 0)
DATA_REG_ADDR = 0x58        # Data register address (R/W = 1, RS2:RS0 = 011)
CONFIG_REG_ADDR = 0x10  # Configuration register address
COMM_READ_CONFIG = COMM_READ | CONFIG_REG_ADDR  # Read from config register
COMM_WRITE_CONFIG = COMM_WRITE | CONFIG_REG_ADDR  # Write to config register
MODE_REG_ADDR = 0x08
STATUS_REG_ADDR = 0x00
COMM_WRITE_MODE = COMM_WRITE | MODE_REG_ADDR

def set_channel(channel):
    if not (1 <= channel <= 3):
        print("Choose proper channel")
        return
    
    channel_map = {
        1: 0x1,
        2: 0x2,
        3: 0x4
    }

    config_value = read_configuration_register()

    print("Current config:"+str(config_value))
    print(f"{config_value:024b}")

    config_value &= (0xFF00FF) #clear channel
    config_value |= (channel_map[channel]<<8)

    config_bytes = [
        COMM_WRITE_CONFIG,  # Write command to config register
        (config_value >> 16) & 0xFF,  # MSB
        (config_value >> 8) & 0xFF,   # Middle byte
        config_value & 0xFF,          # LSB
    ]

    print("New config:"+str(config_value))
    print(f"{config_value:024b}")

    spi.xfer3(config_bytes,1)

    return

def read_configuration_register():
    # Command to read the configuration register
    response = spi.xfer3([0x50],3)
    config_value = (response[0] << 16) | (response[1] << 8) | response[2]
    return config_value

#reset first
spi.xfer3(RESET_COMMAND,0)

time.sleep(0.1)

set_channel(2)

time.sleep(0.1)

#read data register
readVals = spi.xfer3([DATA_REG_ADDR], 3)

adcVal = (readVals[0] << 16) | (readVals[1] << 8) | readVals[2]

print("ADC:", adcVal)

spi.close()
```

This the console output:
```
root@Omega-E58B:~# python3 spi3.py
Current config:279
000000000000000100010111  <-- this is proper reading
New config:535
000000000000001000010111  <-- this is proper setting for switching to channel 2
ADC: 8387985 <-- this is reading from channel 1
```
I see that channel is not switched althought proper modificiation was made to config_bytes which were sent in set_channel(2).
Therefore the ADC reading is also wrong because it is from channel 1 not 2.

If I repeat for instance to call function read_configuration _register() two times the first time I get proper reading and second time I get only only 1 if I print bits.

I am not sure it is a problem because spi mode is not properly set or there is something else wrong in the spidev or my program. But the same program works on RPI 5 by using spidev and xfer2 function.

If I change:
for spi.open(0,1) to
spi.open(1,0) to use software SPI bus I get this:
```
root@Omega-E58B:~# python3 spi3.py
Current config:16777215
111111111111111111111111
New config:16712447
111111110000001011111111
ADC: 16777215
```
and if I enable spi.mode=3 I get the same result. I suppose this is because of wrong wiring mentioned at the begginig.

greenbreakfast, 2025-02-06:
@denfaj Yes, I imagine you'll get random values if you can't access all of the GPIOs used by the software SPI bus.

We've made a new sw spi dt overlay package called `onion-dt-overlay-sw-spi-alt` that sets up a software SPI bus on more easily accessible GPIOs:
sck -> gpio 3
cs0 -> gpio 2
mosi -> gpio 1
miso -> gpio 0

Please:
1. Uninstall the existing sw spi package by running `opkg remove onion-dt-overlay-sw-spi`
2. reboot your device
3. install the new `onion-dt-overlay-sw-spi-alt` package
4. Try running your program again 

Let me know how it goes