OnionIoT/OpenWRT-Packages #112: Cannot connect to Omega 2p after upgrade to 23x firmware

castaway, 2025-05-02:
# What package does this bug affect?

23.05.3-20250205

# **Describe the bug**

SSH into newly installed 23.x firmware does not connect, the AP is up, but ssh times out.

# **To Reproduce**

Followed steps on https://documentation.onioniot.com/firmware/installing-firmware/ to upgrade to 23x from existing firmware, using webgui on previous install.

Cannot connect via Serial Command Line, as I only have an Omega 2+ and a Power Dock (no usb serial chip).

Tried ssh as suggested by @greenbreakfast, using my Android phone + termux. Phone is connected to omega-XXXX AP, has been given IP 192.168.3.175 (shown in termux), yet ssh root@192.168.3.1 times out.

# **Expected behavior**
ssh to connect, take password, show me busybox

# **Screenshots or Console output**
[image]

[image]

# **Version Info (please complete the following information):**
 - Firmware version: 23.05.3-20250205
 - Repo branch (if applicable) release

# **Additional context**

greenbreakfast, 2025-05-02:
@castaway thanks for submitting a detailed issue.

Can you please try using a mac, windows, or linux computer to see if SSH works on those platforms? 
Did the SSH to your Android phone + termux work with the previous v0.3.4 firmware?

castaway, 2025-05-03:
Somehow this didn't occur to me until last night either (sitting at my desktop I only have my phone available), can confirm it DOES work, from my Chromebook (which seems odd as the linux on that surely goes through more network links than my phone app does)..

greenbreakfast, 2025-05-06:
> can confirm it DOES work, from my Chromebook

@castaway Glad to hear you can successfully SSH from a computer.

> I don;t think I ever tried termux with the previous firmware, as I went to the webui, set it up with the local wifi creds, then used my desktop to talk to it - no need for an extra device then

Let me know the result if you do end up trying out the previous v0.3.4 firmware with termux! In the meantime I will close this issue. Feel free to reopen it (or make a new issue) if anything else comes up.